code-iai/ROSIntegration #1: Which is the version of Unreal Engine that this project works with?

rubenalvesbcn, 2017-11-06:
Hi all,

I've tried to use this project on `Unreal 4.14.3 ` and on `Unreal 4.18.0` and on both cases I had problems (with Linux Ubuntu 16.04):

1. [ReadOnlyTargetRules](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/ROSIntegration.Build.cs#L23) is not recognized, so I had to rename it to `TargetInfo`.
2. [: base(Target)](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/ROSIntegration.Build.cs#L23) also raises the error `The type 'UnrealBuildTool.ModuleRules' does not contain a constructor that takes '1' arguments`, then I had to remove that part.
3. I had to remove the line that contains [UseExplicitOrSharedPCHs](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/ROSIntegration.Build.cs#L23) because of the error ` 'UnrealBuildTool.ModuleRules.PCHUsageMode' does not contain a definition for 'UseExplicitOrSharedPCHs'`
4. Unreal complains that [MyFancyPluginObject.h](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/Classes/MyFancyPluginObject.h) must be included on all `.cpp` files as we can see below:
```
ERROR: All source files in module "ROSIntegration" must include the same precompiled header first.  Currently "/home/me/unreal_ws/MyProject/Plugins/ROSIntegration/Source/ROSIntegration/Classes/MyFancyPluginObject.h" is included by most of the source files.  The following source files are not including "/home/me/unreal_ws/MyProject/Plugins/ROSIntegration/Source/ROSIntegration/Classes/MyFancyPluginObject.h" as their first include:

here we have a list of all .cpp files on the ROSIntegration on the project
```

As you can see, I have a bunch of errors when trying to use this plugin when trying it with `Unreal 4.14.3` and `Unreal 4.18.0`, so, is there a specific Unreal version or specific setup required to that makes this plugin work?

Sanic, 2017-11-06:
Hi Ruben,

thanks for pointing out the missing UE4 version statement on the README. 
I'll add this as soon as i have access to my setup again.
Please try UE4 4.16 and see if that works.
4.14 is definitely not supported, because between 4.14 and 4.16 the API of the build system has changed which explains some of the errors in your text.
I've not yet tested the Plugin on 4.18 but will do so in the future.

rubenalvesbcn, 2017-11-06:
Hi @Sanic,

I see that I made a mistake.

I thought that I was testing with `Unreal 4.14.3` and later `Unreal 4.18.0` but I was testing actually only with `Unreal 4.14.3` on both tests.

I'll test it again with `v4.18.0` and see what happens.

Sorry for the noise here :)

Sanic, 2017-11-07:
Hi @rubenalvesbcn 
The Unreal version description has been added to the README of this project.
I'll close this issue now, but please feel free to open new issues for other upcoming problems.

closed 2017-11-07 by Sanic