code-iai/ROSIntegration #3: Where is the 'AdditionalDependencies' section of my Unreal Project?

aaronsnoswell, 2018-01-09:
Hello!

Thank you for this excellent repo. I'm going to be doing some UE4 development with ROS in the near future so you might see some issues (and hopefully pull requests) from me soon.

In README.md, what does "Add ‚ROSIntegration’ to your AdditionalDependencies Section of your Unreal Project" refer to? I can't figure out what this step is saying I should do. Could you elaborate?

Sanic, 2018-01-10:
Hi @aaronsnoswell and thank you very much for your support on this project.

The README is mentioning the .uproject file that is in the root folder of your Unreal project. 
Please see the following code snippet:

           .....
           "Type": "Runtime",
           "LoadingPhase": "Default",
           "AdditionalDependencies": [
               "Engine",
               "ROSIntegration"
           ]
           ....

Have you been able to use the plugin without this dependency configuration?
iirc i needed this configuration to link between unreal plugins, but i was able to clean and compile my Unreal project now without the dependencies.

aaronsnoswell, 2018-01-10:
Hi @Sanic No problems - this is a great library :)

Thank you for the explanation. Indeed - I am able to use the plugin just fine without updating my `.uproject`. Instead, I went into Edit> Plugins > Project and enabled the ROSIntegration plugin there. My full `.uproject` is below for reference (ignore the SteamVR bit).

```
{
	"FileVersion": 3,
	"EngineAssociation": "4.18",
	"Category": "",
	"Description": "",
	"Modules": [
		{
			"Name": "TestROSIntegration",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"Engine"
			]
		}
	],
	"Plugins": [
		{
			"Name": "SteamVR",
			"Enabled": false
		}
	]
}
```

closed 2018-01-10 by aaronsnoswell

Sanic, 2018-01-17:
Just a note on this manner:
When you want to build the Plugin on Linux you need to specify atleast a:
  PublicDependencyModuleNames.AddRange(new string[] { "ROSIntegration" });
in order to get the necessary includes.
I'll test this on windows and see if this has some weird side-effects there.