code-iai/ROSIntegration #8: repo name confusing

daniel86, 2018-01-09:
it should be something like "unreal_ros_integration" instead

Hemofektik, 2018-03-20:
What convention do you mean? I personally would like to omit "unreal" from the name because it would be redundant as a plugin that only works with unreal.

daniel86, 2018-03-20:
the rather implicit conventions can be derived by looking at how repositories are named in the code-iai group -- which is lower case with underscore.

I would add the hint that it is unreal related in the name of the repository to avoid confusion about what is integrated with ROS -- the IAI is integrating lots of stuff with ROS ;)

Hemofektik, 2018-03-20:
The current naming is aligned with the Unreal Naming Convention which is nice, but I understand your concern as well. It's probably a matter of opinion in the end.

daniel86, 2018-03-21:
Well why don't you create then a "UnrealRobots" group or so? Then everything in this group can be named according to Unreal conventions.