code-iai/ROSIntegration #11: Add instructions on adding custom message types

aaronsnoswell, 2018-01-11:
It currently requires a fair bit of effort to add message types to the core plugin. The readme should include a section showing how end-users can add new message definitions to their projects. I'll try and add this in the next week or so.

oscarmendezm, 2018-01-19:
Hi, I am trying to add some functionality to do ROS->UE4 geometry_msgs. Do you have a quick and dirty explanation of what needs to be done?

aaronsnoswell, 2018-01-23:
Ok @oscarmendezm I've pushed the branch I was referring to. Please see [support-point-clouds](https://github.com/aaronsnoswell/ROSIntegration/tree/support-point-clouds) for examples of how to add custom point clouds. I'll update the readme at some point with some clearer instructions.

Hemofektik, 2018-11-19:
This was fixed by the latest PR #51 right?