code-iai/ROSIntegration #12: Access violation in Topic->Publish

aaronsnoswell, 2018-01-11:
I've seen an access violation crash in the code for Topic->Publish that seems to be a race condition. So far I've seen it under two circumstances;

 1. The server IP details are incorrect, but I try to publish something anyway
 2. The server IP details are correct, I publish something, and then the remote host disconnects for some reason (e.g. network interrupted or remote node killed).

I'll add more details as I look into this over the coming week or two.

Sanic, 2018-03-20:
Hi @Hemofektik 
Is this issue somehow related to your latest fixes? In 9f7468e3889b73b29a46d4f1e0ea7d6f02a93012 you referenced some premature tcp disconnects and introduced locking operations for the callback topics which brought me back to this issue....

Hemofektik, 2018-03-20:
Hi,

the premature disconnects and the locking operation was for incoming messages only. But the problems fixed by that sound exactly as described by @aaronsnoswell but not for publishing messages.
Race conditions with topic->publish could only happen if you use one or multiple topics from different threads. That is not supported yet. @aaronsnoswell Do you use multiple threads.

P.S.: I already checked in queued publishers that would allow this behavior to work (additionally to the main feature to send messages asynchronously to the rosbridge and queue the messages). I will make a pull request for that soon.

Hemofektik, 2018-03-20:
Please check if #23 fixes your problem.

aaronsnoswell, 2018-03-22:
@Hemofektik No I don't believe I was using multiple threads. Sorry I don't have my project where I was testing this library available any more so I'm unable to test :(

Hemofektik, 2018-03-22:
I just read your initial post again and I think what you experienced was the GraceTime of the TCPConnection. It tries to send the message forcefully over a period of 4 seconds before it gives up and discards the message. This blocks the main thread for 4 seconds. This will happen for each message send as soon as the connection to the rosbridge_server is lost.

IMHO, this GraceTime thingy has to be removed anyway because this type of error handling is way too deep buried into the Plugin and nobody is able anymore to decide by themselves to handle such situations differently.

Hemofektik, 2018-11-21:
The handling of ROS Connections/sockets/publishing was refactored very much since back then. This problem most probably no longer exists.

Sanic, 2018-11-21:
Thanks for the info. Then i'll close this issue. For future readers: Please don't hesitate to open new tickets when the problem appears again. Please reference this issue when you see this problem again.

closed 2018-11-21 by Sanic

finger563, 2020-04-28:
I'm seeing this on the latest codebase (with the required fixes to run on UE 4.24). I'm publishing `TransformedStamped` data (multiple transforms) each @ about 2 Hz. After a minute or two UE4 crashes and points to where I call `TransformTopic->Publish( ... )`  Any ideas why I might be seeing this? the network connection is fine (windows host running UE4 to Linux VM in VMWare) and up until the crash I'm getting all my data just fine.  Sometimes the editor doesn't crash, but the publishers stop sending data (e.g. the publish code is still running in UE4, and rosbridge still shows that the client is connected, but no topics are received anymore).  I think the two issues are related.