code-iai/ROSIntegration #17: Support for UE 4.16.x?

oscarmendezm, 2018-01-22:
Hi,

 I noticed support for 4.16.x was removed. Is there any reason for this? Can I revert back to a version that has it? 

I need to be able to use it with some existing projects that would be a pain to upgrade.

Thanks!

oscarmendezm, 2018-01-22:
Hi Sanic, 

Thanks for the quick reply! I've tried this on UE4.16.3 on Windows 10. I get the following errors:

[image]
[image]

I am still getting to grips with UE, so I am not sure how to fix this.

Thanks

Sanic, 2018-01-22:
Hi!
The first warning is normal if you've added the Plugin to your Unreal project, but haven't it build yet. The second box is the interesting one, but you and we need more details.
Please check how you can access your compile logs to see the actual error.

oscarmendezm, 2018-01-22:
This is the log file I get when I make a new project and try to start it. 

[ROSTEST_v16.log](https://github.com/code-iai/ROSIntegration/files/1652314/ROSTEST_v16.log)

I can't seem to find any reference to your code, instead it complains about some other stuff. However, if I remove the plugin the project starts without any issues.

Sanic, 2018-01-24:
Hi @oscarmendezm 
Please have a look at https://github.com/code-iai/ROSIntegration/issues/19
Did you use a Blueprint project in your previous tests?
Can your issue be solved when you start with a fresh C++ Unreal Project?

closed 2018-03-20 by Sanic