code-iai/ROSIntegration #18: TCPConnection: Handle Joined Messsages

aaronsnoswell, 2018-01-23:
Hi Sanic,

I'm trying to implement support for [`sensor_msgs/PointCloud2`](http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud2.html) (and friends) so I can visualize point clouds in UE4. See my branch [support-point-clouds](https://github.com/aaronsnoswell/ROSIntegration/tree/support-point-clouds).

So far, I have successfully added the message definitions for the required message types. I have a Ubuntu 16.04 ROS kinetic instance running ROSBridge in BSON mode, and I'm publishing a topic that is a throttled sensor_msgs/PointCloud2 (about 1 frame every 2 seconds). I can visualize the PointCloud2 in RVIZ just fine, and I know my ROS <> UE4 connection is fine, because I can subscribe / publish `std_msgs/String` topics without error.

When I try to subscribe to the `sensor_msgs/PointCloud2` topic, the BSON parsing fails at [`bson_init_static()` circa line 254 in TCPConnection.cpp](https://github.com/aaronsnoswell/ROSIntegration/blob/b626d98ca6260dad7ce89e1931bb2a9b9186f3a6/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L254). I believe this is happening because of a comment [circa line 140 in TCPConnection](https://github.com/aaronsnoswell/ROSIntegration/blob/b626d98ca6260dad7ce89e1931bb2a9b9186f3a6/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L140) - it says

```
// TODO handle joined messages while reading the buffer
uint32 count;
```

After changing the `std::cout` calls in `TCPConnection.cpp` to `UE_LOG` calls, I get the following console output in Unreal;

```
LogTemp: Repeating last play command: Selected Viewport
LogPlayLevel: PlayLevel: No blueprints needed recompiling
PIE: New page: PIE session: Default (23 Jan 2018 9:43:24 pm)
LogPlayLevel: Creating play world package: /Game/UEDPIE_0_Default
LogPlayLevel: PIE: StaticDuplicateObject took: (0.026701s)
LogAIModule: Creating AISystem for world Default
LogPlayLevel: PIE: World Init took: (0.003329s)
LogPlayLevel: PIE: Created PIE world by copying editor world from /Game/Default.Default to /Game/UEDPIE_0_Default.Default (0.031225s)
LogTemp: Setting up receiver thread...
LogTemp: Receiving
LogTemp: rosbridge2cpp init successful
LogTemp: Send 201 bytes from (201 / 201) Bytes 
LogTemp: T
LogTemp: <SendDataEnd:53641>
LogTemp: Send 218 bytes from (218 / 218) Bytes 
LogTemp: T
LogTemp: <SendDataEnd:8110>
LogInit: XAudio2 using 'Speaker/HP (Realtek High Definition Audio)' : 2 channels at 48 kHz using 32 bits per sample (channel mask 0x3)
LogInit: FAudioDevice initialized.
LogLoad: Game class is 'GameModeBase'
LogWorld: Bringing World /Game/UEDPIE_0_Default.Default up for play (max tick rate 60) at 2018.01.23-11.43.24
LogWorld: Bringing up level for play took: 0.001184
LogTemp: Added Default__GeometryMsgsPoint32Converter with type geometry_msgs/Point32 to TopicConverterMap
LogTemp: Added Default__GeometryMsgsQuaternionConverter with type geometry_msgs/Quaternion to TopicConverterMap
LogTemp: Added Default__GeometryMsgsTransformConverter with type geometry_msgs/Transform to TopicConverterMap
LogTemp: Added Default__GeometryMsgsTransformStampedConverter with type geometry_msgs/TransformStamped to TopicConverterMap
LogTemp: Added Default__GeometryMsgsVector3Converter with type geometry_msgs/Vector3 to TopicConverterMap
LogTemp: Added Default__SensorMsgsCameraInfoConverter with type sensor_msgs/CameraInfo to TopicConverterMap
LogTemp: Added Default__SensorMsgsChannelFloat32Converter with type sensor_msgs/ChannelFloat32 to TopicConverterMap
LogTemp: Added Default__SensorMsgsImageConverter with type sensor_msgs/Image to TopicConverterMap
LogTemp: Added Default__SensorMsgsPointCloud2Converter with type sensor_msgs/PointCloud2 to TopicConverterMap
LogTemp: Added Default__SensorMsgsPointCloudConverter with type sensor_msgs/PointCloud to TopicConverterMap
LogTemp: Added Default__SensorMsgsPointFieldConverter with type sensor_msgs/PointField to TopicConverterMap
LogTemp: Added Default__SensorMsgsRegionOfInterestConverter with type sensor_msgs/RegionOfInterest to TopicConverterMap
LogTemp: Added Default__StdMsgsHeaderConverter with type std_msgs/Header to TopicConverterMap
LogTemp: Added Default__StdMsgsStringConverter with type std_msgs/String to TopicConverterMap
LogTemp: Added Default__Tf2MsgsTFMessageConverter with type tf2_msgs/TFMessage to TopicConverterMap
LogTemp: Send 183 bytes from (183 / 183) Bytes 
LogTemp: T
LogTemp: <SendDataEnd:15088>
LogTemp: Subscribed now
PIE: Play in editor start time for /Game/UEDPIE_0_Default 0.628
LogTemp: d14600
LogTemp: Reading length...
LogTemp: Total message length is: 6947308
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Error on BSON parse - Ignoring message
LogTemp: d65536
LogTemp: Reading length...
LogTemp: Total message length is: 6947308
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Error on BSON parse - Ignoring message
[... 1129 lines cut ...]
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: d65536
LogTemp: Message read mode
LogTemp: Error: Failed to recv() in message retreival mode even though data is pending. Count vs. bytes_read: 65536, 0
LogTemp: Error: Error on connection
```

Do you think this error could be because 'handle joined messages while reading the buffer' hasn't been implemented yet? What would it take to implement this? If it would make it easier for you to assist me, I can share a bagfile that publishes a simple `sensor_msgs/PointCloud2` topic for testing purposes.

Thank you,

Aaron.

aaronsnoswell, 2018-01-23:
It might be easier to get the gist of the UE4 log in an image;

[image]

aaronsnoswell, 2018-01-23:
Another update here: I updated my ROS nodes so that I'm downsampling the point cloud massively (see https://github.com/aaronsnoswell/ue4_comms_test/blob/master/src/pcl_downsampler.cpp). When I do this, I am able to recieve the `sensor_msgs/PointCloud2` messages fine. As such, this definitely seems to be a problem with receiving large bson messages that are split over multiple TCP packets.

Can we work out an approach to fixing this problem? I'm happy to do the work if you advise what needs to be done :) I'm not super familiar with the code in `TCPConnection.cpp`

Sanic, 2018-01-30:
Hi @aaronsnoswell 
Thanks for opening this issue and sorry for the delay.
This problem might be indeed a problem with the TCPConnection class.
I think the relevant part starts at https://github.com/code-iai/ROSIntegration/blob/7e1716c4098693aef746165f17531e9cd0265ed7/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L205 . The main idea is to begin by reading the first 4 bytes. These bytes represent the length field of a bson packet: http://bsonspec.org/spec.html. By reading this length, we can instruct the next recv() calls to receive exactly that many bytes.
After these 4 four bytes have been read, bson_state_read_length will be false. This will cause the jump to https://github.com/code-iai/ROSIntegration/blob/7e1716c4098693aef746165f17531e9cd0265ed7/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L237 to handle the actual message reception. I guess the problems begin there. Maybe `if( _sock->Recv(binary_temp.GetData(), bson_msg_length - 4, bytes_read) )` can return less then the requested bson_msg_length - 4 bytes on fragmented packets. So one would have to reiterate to collect more data in the buffer. 
Additionally, i'm not 100% sure if `if(msg_size_in_buffer == bson_msg_length)` is correct. msg_size_in_buffer is calculated by calling Num() on a TArray. However, this array is resized with SetNumUninitialized() : https://github.com/code-iai/ROSIntegration/blob/7e1716c4098693aef746165f17531e9cd0265ed7/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L245
So Num() may not count the actual elements in the list, but rather the received elements + uninitialized elements.

TaylorDale, 2021-05-14:
Has this been fixed now? I am just reviewing the code for TCPConnection.cpp and it looks substantially different to what it did in 2018, and appears that it will correctly wait to fill a buffer based on the 4 byte BSON prefix. 

However we are still receiving BSON decode issues with our pointcloud frames ("Error on BSON parse - Ignoring message"), using v0.11.10 of rosbridge server. Any clue if this could be related to it?

PetterVMC, 2023-07-31:
@Sanic 
> . The main idea is to begin by reading the first 4 bytes. These bytes represent the length field of a bson packet: http://bsonspec.org/spec.html. By reading this length, we can instruct the next [...]

About a year ago I had this exact problem you're describing Sanic, so I modified TCPConnection.cpp to be able to read headers from fragmented packets. I have a pending pull request #179 on this . Would appreciate if someone could have a look at it and accept it if it seems OK. I've been using my modified version multiple times now over the last year without any apparent problem.