code-iai/ROSIntegration #19: Error when opening a project using this plugin

samcaulfield, 2018-01-24:
Hi,

I am receiving this error when I start a project that uses this plugin:
[image]

Clicking "No" results in this error:
[image]

My setup is Ubuntu 16.04.3 with Unreal Engine 4.18.2.

I set up Unreal Engine using the following steps:
```
git clone https://github.com/EpicGames/UnrealEngine.git
cd UnrealEngine
git checkout "4.18.2-release"
./Setup.sh
./GenerateProjectFiles.sh
make
cd Engine/Binaries/Linux
./UE4Editor
<Created a project called ROSIntegrationTest then exited the editor>
```

I set up this plugin using the following steps:
```
cd ~/Documents/Unreal\ Projects/ROSIntegrationTest
mkdir Plugins
cd Plugins
git clone https://github.com/code-iai/ROSIntegration.git
cd ~/UnrealEngine/Engine/Binaries/Linux
./UE4Editor
```

Here is the [UE4.log](https://github.com/code-iai/ROSIntegration/files/1660318/UE4.log) generated by this attempt.

I'm new to Unreal so perhaps I'm making a rudimentary mistake. Any help you could provide would be appreciated.

Thanks

Sanic, 2018-01-24:
Hi!
Thanks for the bug report.
This might be related to https://github.com/code-iai/ROSIntegration/issues/17.

I'm investigating the issue right now by installing the plugin and compiling it from scratch.

However, can you please tell me which kind of Unreal Project you've created?
A C++ or a Blueprint version?

samcaulfield, 2018-01-24:
The project I created was a Blueprint-based one using the "First Person" example.

I just created a C++ version of the "First Person" example and it seems to work fine. The editor starts up and I can see the plugin in the plugins list.

Sanic, 2018-01-24:
Hi @samcaulfield 

Thanks for the report. I was also just testing this and i'm suspecting that the Plugin will not be compiled because a blank Blueprint project will not invoke the build process.
iirc, you can also start with a Blueprint project, but i guess you have to have atleast one C++ class. This will instruct the build process to compile the whole project which should also fetch the code of the plugin.

closed 2018-01-24 by Sanic

samcaulfield, 2018-01-24:
OK. Thanks for looking into it and the explanation of the cause of the problem.