code-iai/ROSIntegration #25: Rebuild from Source...

gonzaloaguilarjimenez, 2018-04-21:
Good afternoon again, 

I write you because some days ago I had the ROSIntegration and ROSIntegrationVision totally installed but due to some failure on my project I had to delete and create a new onw.

Now I am trying to install ROSIntegration and I cannot because I get the following failure. 

[image]

For sure, I click YES and the message is as follows: 

[image]

I try to rebuild and generate the project files from source but no good results.

Could you please help me? 

I have seen that more people had this issue but using another UE4 version.

Thanks a lot.

Best regards.

Hemofektik, 2018-04-21:
Hi, which version of UE4 do you use? Additionally the generation of project files output logs that could help to identify the issue.

yskim041, 2018-04-29:
It's because there are some declarations of `UPROPERTY` without any explicit Category specifier. In this case, when you build from source, you will be able to find error messages like this:
```
LogCompile: Error: An explicit Category specifier is required for any property exposed to the editor or Blueprints in an Engine module.
```

yskim041, 2018-04-29:
If you want to build from soure on Linux, you need to fix the declarations of `UPROPERTY` that are exposed to the editor or blueprints without any explicit Category specifier.

For example,
https://github.com/code-iai/ROSIntegration/blob/7376006abc8811454fddc1d8609ffd33f34a538e/Source/ROSIntegration/Classes/TFBroadcastComponent.h#L34
should be like
```
UPROPERTY(EditAnywhere, Category = "explicit specifier")
```

Hemofektik, 2018-05-15:
It should compile again on linux. Keep in mind that this plugin is not an Engine plugin and should be build with your custom project and not put into the Engine/Plugins Folder.

closed 2018-05-23 by Sanic