code-iai/ROSIntegration #28: rosbridge2cpp compile errors when building plugin

jacobdeery, 2018-05-09:
I'm using Unreal 4.18.3 on Ubuntu 16.04. I've done the following:

1. Create a new C++ Unreal project.
2. Clone this repo into the `<project>/Plugins` directory.
3. Enable the plugin through the editor menu.
4. Restart the editor in order to build the plugin.

The build fails during the restart. Looking at the log, I found the following relevant errors/warnings:
[unreal_build_errors.txt](https://github.com/code-iai/ROSIntegration/files/1988202/unreal_build_errors.txt)

jacobdeery, 2018-05-09:
This seems to be a C++11 issue, from what I can tell. `std::function::target` was only introduced in C++11, and isn't described as one of the C++11 features that Unreal supports ([here](https://docs.unrealengine.com/en-us/Programming/Development/CodingStandard#c++11andmodernlanguagesyntax)).

Hemofektik, 2018-05-10:
I did not test the latest changes on linux yet, sorry. I will try to fix that as soon as I can.

dawgster, 2018-05-11:
A fix would be greatly appreciated! Thanks

robbeofficial, 2018-05-11:
same error on ubuntu 18.05 / Unreal 4.19.1 (clang 5.0.1)

Hemofektik, 2018-05-13:
Hi, as soon as the PR #29 is merged, it should work for you again. Sorry for the inconvenience.

Hemofektik, 2018-05-15:
Hi, please verify whether this is fixed for you!

jacobdeery, 2018-05-15:
Compiles correctly now, thanks!

Sanic, 2018-05-23:
Thanks @Hemofektik for fixing this in your PR.

closed 2018-05-23 by Sanic