code-iai/ROSIntegration #30: Not communicating with rosbridge

BuggeTHK, 2018-05-18:
Hi,
I've set the plugin and world up as per the instructions, and on play rosbridge registers a connection. 
But putting the example publishing code in an actor, rosbridge registers no communication. Is there something in particular I have to do?
Thanks. 
This is what I have in beginplay:
UTopic *ExampleTopic = NewObject<UTopic>(UTopic::StaticClass());
	UROSIntegrationGameInstance* ROSInstance = Cast<UROSIntegrationGameInstance>(GetGameInstance());
	ExampleTopic->Init(ROSInstance->ROSIntegrationCore, TEXT("/chatter"), TEXT("std_msgs/String"));

	// (Optional) Advertise the topic
	// Currently unimplemented in the plugin
	//ExamplePublishTopic->Advertise();

	// Publish a string to the topic
	TSharedPtr<ROSMessages::std_msgs::String> StringMessage(new ROSMessages::std_msgs::String("This is an example"));
	ExampleTopic->Publish(StringMessage);

Hemofektik, 2018-05-18:
looks good so far. Did you start the rosbridge as suggested? (roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True)
How exactly did you verify that nothing was received by the rosbridge?

BuggeTHK, 2018-05-18:
I've tried both `roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True`
as well as `rosrun rosbridge_server rosbridge_tcp.py --bson_only_mode. `
It's the newest rosbridge version available for indigo, taken from the RobotWebTools/rosbridge_suite repository. (version 0.8.6)
I think rosbridge is supposed to give me an error if the data received is broken in some way, but it doesn't seem to give me anything (besides confirming a connection has been made). 

I've also tried making a blueprint to subscribe to a topic, without success. 
The subscribe function returns true even if it tries a topic which doesn't exist, is that supposed to happen?

Thanks for helping me out.

Hemofektik, 2018-05-18:
The rosbridge would output an error if it gets any corrupted data messages that is true. But if the data is fine it would not output anything on the console. It would just republish the message to the ROS Environment. You need at least have a console open with "rostopic echo /chatter" before starting your BeginPlay to see if the message really is dropped somewhere in the middle.

Subscribing to a topic that was not advertised yet is completely fine in the ROS world because it is not safe to say when one ROS Node is started or available before any other node. Everything can come and go. As soon as someone is publishing to this topic your subscription should work then without issues.

BuggeTHK, 2018-05-18:
I wasn't sure if I needed to manually create the topic (since the plugin doesn't support advertising, as far as I understand), so I made a little python script that continually publishes to /chatter.  
"rostopic echo /chatter" picks up these messages, but nothing seems to appear from my Unreal computer. 

The BP with subscribe isn't picking up anything either.  If I subscribe to a string topic, "Event On String Message" is supposed to fire every time something is published to the topic, right?

Hemofektik, 2018-05-18:
Yes the BP should fire every time. Sounds as you set it up correctly. I have no idea at the moment what the issue could be. The next thing I would do is to debug the code of the rosbridge2cpp code of the ROSIntegration Plugin to see what is happening with the messages.

BuggeTHK, 2018-05-18:
Hmm. This will probably take me some time. 

I looked at the value of "ExampleTopic" during execution, which gave me this: 
```
-		ExampleTopic	0x000001fbbc8e7f40 (Name=0x000001fb8471aea8 "Topic"_1)	UTopic * {UE4Editor-ROSIntegration.dll!UTopic}
+		[UTopic]	(Name=0x000001fb8471aea8 "Topic"_1)	UE4Editor-ROSIntegration.dll!UTopic
+		UObject	(Name=0x000001fb8471aea8 "Topic"_1)	UObject
-		_State	{Connected=true Advertised=false Subscribed=false ...}	UTopic::State
		Connected	true	bool
		Advertised	false	bool
		Subscribed	false	bool
		Blueprint	false	bool
		BlueprintMessageType	String (0 '\0')	EMessageType
		_Implementation	0x000001fbb36b7900 {...}	UTopic::Impl *
```
I see that Connected is true. Does that narrow anything down?

Hemofektik, 2018-05-18:
Looks good. You may have to go deeper to see what is really going on. You could try to set a breakpoint at line 422 of rosbridge2cpp/ros_bridge.cpp and see if the SendMessage call returns success. If so, the message is lost somewhere on the way to the rosbridge node or the republishing of the message into the ROS realm done by the rosbridge node is somehow faulty.

On the message receiving side you could check rosbridge2cpp/TCPConnection.cpp from line 168 on whether the data that is incoming is parsed correctly and a few lines down passed to the incoming_message_callback_bson_().

BuggeTHK, 2018-05-18:
Thank you!
success = true, so I guess the message is lost underway. Does this mean I should stop looking in the ROSIntegration plugin, and start troubleshooting rosbridge?

Hemofektik, 2018-05-18:
Yes, it's weird that the initial connection works and sending the data also works but the rosbridge is not communicating with the ROS. You said Unreal **Computer**. May be you started the rosbridge on the computer that is also running unreal and also started a ROS Core on another computer? starting the RosBridge with roslaunch you get two ROS cores, one on each computer.  Or is the ROSbridge run on the other machine where your python script also runs?

BuggeTHK, 2018-05-18:
My setup is a Windows 10 computer with Unreal installed, with a direct wired connection to a computer running Ubuntu 16.04 and ROS Indigo. ROSbridge and the python script run on this computer.  
Yes, it might have been easier if rosbridge had refused to run, or at least complained a bit more. 

This is all I get (I'm sure this isn't very helpful, though):
```
$ rosrun rosbridge_server rosbridge_tcp.py --bson_only_
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
trying to start rosbridge TCP server..

[INFO] [WallTime: 1526627547.316517] Rosbridge TCP server started on port 9090
[INFO] [WallTime: 1526628606.244454] [Client 0] connected. 1 client total.
[INFO] [WallTime: 1526628688.811291] [Client 0] disconnected. 0 client total.
[INFO] [WallTime: 1526628908.009053] [Client 1] connected. 1 client total.
[INFO] [WallTime: 1526628914.533406] [Client 1] disconnected. 0 client total.
[INFO] [WallTime: 1526628988.462920] [Client 2] connected. 1 client total.
[INFO] [WallTime: 1526629007.630282] [Client 2] disconnected. 0 client total.
[INFO] [WallTime: 1526629023.261065] [Client 3] connected. 1 client total.
[INFO] [WallTime: 1526629034.406234] [Client 3] disconnected. 0 client total.
[INFO] [WallTime: 1526629074.830700] [Client 4] connected. 1 client total.
[INFO] [WallTime: 1526629086.669768] [Client 4] disconnected. 0 client total.
[INFO] [WallTime: 1526629530.948201] [Client 5] connected. 1 client total.
[INFO] [WallTime: 1526629539.277323] [Client 5] disconnected. 0 client total.
[INFO] [WallTime: 1526629918.764458] [Client 6] connected. 1 client total.
[INFO] [WallTime: 1526629946.174431] [Client 6] disconnected. 0 client total.
[INFO] [WallTime: 1526630039.443568] [Client 7] connected. 1 client total.
[INFO] [WallTime: 1526630229.181108] [Client 7] disconnected. 0 client total.
[INFO] [WallTime: 1526630229.889319] [Client 8] connected. 1 client total.
[INFO] [WallTime: 1526630592.367415] [Client 8] disconnected. 0 client total.
[INFO] [WallTime: 1526630865.160718] [Client 9] connected. 1 client total.
[INFO] [WallTime: 1526630868.073090] [Client 9] disconnected. 0 client total.
[INFO] [WallTime: 1526630962.409896] [Client 10] connected. 1 client total.
[INFO] [WallTime: 1526631009.869886] [Client 10] disconnected. 0 client total.
```

Hemofektik, 2018-05-18:
Interesting. The BP that subscribes to a topic should result in an output at the rosbridge as something like the following:
`[INFO] [1526630223.419235]: [Client 0] Subscribed to /chatter`

Are you sure you subscribing at event beginplay to the topic via the BP?

Other than that it looks fine if you only have the topic that publishes from Unreal via code in Beginplay. I have to admin I never tried to using it like that. Maybe it helps to put the following part:

```
// Publish a string to the topic
	TSharedPtr<ROSMessages::std_msgs::String> StringMessage(new ROSMessages::std_msgs::String("This is an example"));
	ExampleTopic->Publish(StringMessage);
```

Into the Tick function of your Actor instead of BeginPlay, just to be sure this has nothing to do with Setting up the connection in right order prior to sending something.

BuggeTHK, 2018-05-21:
At beginplay, an actor constructs a BP listentotopic, whose "Event on construct" triggers subscribe as well as print  to screen. The blueprint prints to screen, so I think it's subscribing as well. 
Using Wireshark, I captured this packet as the blueprint ran:
```
@.\Ñ¸ûÔ.×.Äg..E.
.rS.@.....À¨..À¨
.3Îç#..[.ÎSÃÍAP.
...ü..J....op...
..unsubscribe..i
d.....subscribe:
/chatter:3..topi
c...../chatter..
```
I also captured these packets when the publishing code runs:
```
@.\Ñ¸ûÔ.×.Äg..E.
..WS@.....À¨..À¨
.3Îò#.ùG|»"bÎ.P.
...'..u....op...
..publish..id...
..publish:/chatt
er:304..topic...
../chatter..latc
h...msg."....dat
a.....This is an
 example...
```
```
@.\Ñ¸ûÔ.×.Äg..E.
.¥1.@.....À¨..À¨
.3Îq#..¼.l.øf.P.
.../..}....op...
..advertise..id.
....advertise:/c
hatter:2325..top
ic...../chatter.
.type.....std_ms
gs/String..queue
_size......latch
...

```
using wireshark on the receiving computer, I saw these same packets arrive. 

I tried putting the publishing string into the tick function, although I had to include the 
```
	UTopic *ExampleTopic = NewObject<UTopic>(UTopic::StaticClass());
	UROSIntegrationGameInstance* ROSInstance = Cast<UROSIntegrationGameInstance>(GetGameInstance());
	ExampleTopic->Init(ROSInstance->ROSIntegrationCore, TEXT("/chatter"), TEXT("std_msgs/String"));

```
to get it to compile. Is it alright to make a UROSIntegrationGameInstance on every tick?

Hemofektik, 2018-05-21:
> Casting to UROSIntegrationGameInstance is not a problem, but you create  new topic every tick. That will not work. Put the ExampleTopic into a member of your class and it will work to put the init into BeginPlay and the rest into the Tick method.

florian-leblanc, 2018-10-26:
I have the same problem here my client who connect disconnect aswell. I'm using roslibjs to connect.
[image]

Sanic, 2018-10-26:
@florian-leblanc We need additional information to debug this.
For what part exactly do you need roslibjs? roslibjs isn't aware of the BSON encoding of this UE4 Plugin and it's also supposed to be used as a client. So i'm not sure how roslibjs comes into play here.

Also: Where do you hold your instance of your Topic? Please note that it should live in some scope that persists: https://github.com/code-iai/ROSIntegration/issues/32

Hemofektik, 2018-10-26:
We also use roslibjs and it works fine. you have to start a rosbridge_websocket on a different port additionally to the rosbridge_tcp though.
I have seen the problem sometimes on windows which runs ROS on Windows Subsystem for Linux. under linux though, it never happened.

florian-leblanc, 2018-10-26:
> @florian-leblanc We need additional information to debug this.
> For what part exactly do you need roslibjs? roslibjs isn't aware of the BSON encoding of this UE4 Plugin and it's also supposed to be [...]

It's ok now i find my problem it was tornado : #353

closed 2018-11-21 by Sanic