code-iai/ROSIntegration #32: An actor subscribed to a topic is automatically unsubscribed after about a minute

jiwoong-choi, 2018-05-25:
First of all, thank you for sharing the great plugin.

I have slightly modified the C++ publish / subscribe examples in the Readme.md.

After setting up & build the plugin, I have created two C++ Actor classes called, Publisher and Subscriber.  In UE4Editor, I have added each class to the level editor by drag & drop.
Basically, the idea is that Publisher publishes a string message at each frame to the example_topic, and Subscriber listens to the example_topic and prints out what it heard to the console.

Every time I run the game in UE4Editor, everything works perfectly for the first 1 minute. However, some point after a minute, UE4Editor crashes with the following error message:

```
.
.
.
Subscriber listened: Message from frame 5035
Subscriber listened: Message from frame 5036
Subscriber listened: Message from frame 5037
Subscriber listened: Message from frame 5039
Subscriber listened: Message from frame 5040
Subscriber listened: Message from frame 5041
Subscriber listened: Message from frame 5042
Subscriber listened: Message from frame 5043
Subscriber listened: Message from frame 5045
Subscriber listened: Message from frame 5046
Subscriber listened: Message from frame 5047
Subscriber listened: Message from frame 5048
Subscriber listened: Message from frame 5049
Subscriber listened: Message from frame 5051
Subscriber listened: Message from frame 5052
Subscriber listened: Message from frame 5053
Subscriber listened: Message from frame 5054
Subscriber listened: Message from frame 5056
Subscriber listened: Message from frame 5057
Subscriber listened: Message from frame 5058
Subscriber listened: Message from frame 5059
Subscriber listened: Message from frame 5060
[ROSBridge] Found CB in UnregisterTopicCallback. Deleting it ... 
[ROSTopic] No callbacks registered anymore - unsubscribe from topic
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
[2018.05.25-09.27.55:775][937]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000e30800a0

[2018.05.25-09.27.55:775][937]LogLinux: Fatal error!

0x00007f9cd4b6a625 FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
0x00007f9cd4a18c75 FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
0x00007f9cd4b1b231 FLinuxCrashContext::CaptureStackTrace()
0x00007f9cca344400 CommonLinuxCrashHandler(FGenericCrashContext const&)
0x00007f9cd4b1d2bd PlatformCrashHandler(int, siginfo_t*, void*)
0x00007f9cd5781390 /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f9cd5781390]
0x00007f9bc6cc3efe rosbridge2cpp::ROSTopic::GeneratePublishID()
0x00007f9bc6cc3048 rosbridge2cpp::ROSTopic::Publish(_bson_t*)
0x00007f9bc6cd7f18 UTopic::Impl::Publish(TSharedPtr<FROSBaseMsg, (ESPMode)0>)
0x00007f9bc6cc98b3 UTopic::Publish(TSharedPtr<FROSBaseMsg, (ESPMode)0>)
0x00007f9bb5d2da58 APublisher::Publish(char const*)
0x00007f9bb5d2d81d APublisher::Tick(float)
0x00007f9cd1243983 FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&)
0x00007f9cd25efab6 FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&)
0x00007f9cd25ef41f TGraphTask<FTickFunctionTask>::ExecuteTask(TArray<FBaseGraphTask*, FDefaultAllocator>&, ENamedThreads::Type)
0x00007f9cd4a4443f FNamedTaskThread::ProcessTasksNamedThread(int, bool)
0x00007f9cd4a43903 FNamedTaskThread::ProcessTasksUntilQuit(int)
0x00007f9cd25e9813 FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool)
0x00007f9cd25e2694 FTickTaskManager::RunTickGroup(ETickingGroup, bool)
0x00007f9cd1c2635f UWorld::RunTickGroup(ETickingGroup, bool)
0x00007f9cd1c303ee UWorld::Tick(ELevelTick, float)
0x00007f9ccb4888a2 UEditorEngine::Tick(float, bool)
0x00007f9ccbeb93e6 UUnrealEdEngine::Tick(float, bool)
0x00000000004255f3 FEngineLoop::Tick() [/home/jchoi/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3339]
[... 224 lines cut ...]
    ExampleTopic = NewObject<UTopic>(UTopic::StaticClass());
    UROSIntegrationGameInstance* rosinst = Cast<UROSIntegrationGameInstance>(GetGameInstance());
    ExampleTopic->Init(rosinst->ROSIntegrationCore, TEXT("/example_topic"), TEXT("std_msgs/String"));

    std::function<void(TSharedPtr<FROSBaseMsg>)> SubscribeCallback = ASubscriber::SubscribeCallbackImpl;
    ExampleTopic->Subscribe(SubscribeCallback);
}

// Called every frame
void ASubscriber::Tick(float DeltaTime)
{
	Super::Tick(DeltaTime);

}

void ASubscriber::SubscribeCallbackImpl(TSharedPtr<FROSBaseMsg> msg)
{
    auto Concrete = StaticCastSharedPtr<ROSMessages::std_msgs::String>(msg);
    if (Concrete.IsValid())
    {
        std::cout << "Subscriber listened: " << TCHAR_TO_UTF8(*(Concrete->_Data)) << std::endl;
    }
    return;
}
```

Hemofektik, 2018-05-25:
Your topics are getting garbage collected and thus unsubscribe automatically.
Put UPROPERTY() before
class UTopic *ExampleTopic;
and it should no longer happen.

jiwoong-choi, 2018-05-28:
Now it works perfectly fine. Thank you so much!

closed 2018-05-28 by Sanic

xgwang, 2020-03-11:
> Your topics are getting garbage collected and thus unsubscribe automatically.
> Put UPROPERTY() before
> class UTopic *ExampleTopic;
> and it should no longer happen.

may i know the GC rules here? why this member pointer is GCed?

thanks.