code-iai/ROSIntegration #35: [UE4.19.2] Build.cs: Definitions -> PublicDefinitions

Dyddye, 2018-06-18:
In Unreal Engine 4.19.2 ```Definitions``` is unknown in line 33 of Build.cs and needs to be replaced with ```PublicDefinitions```.

ROSIntegration/Source/ROSIntegration/ROSIntegration.Build.cs:33
```Definitions.Add("RAPIDJSON_HAS_STDSTRING=1");```
```PublicDefinitions.Add("RAPIDJSON_HAS_STDSTRING=1");```

Sanic, 2018-06-26:
Thanks for letting us know. It seems that there are now some macros to get the UE4 version when the Build file gets compiled:
https://answers.unrealengine.com/questions/671250/check-ue4-version-at-compile-time-c.html
Example: https://github.com/GameAnalytics/GA-SDK-UNREAL/blob/master/GameAnalytics/Source/GameAnalytics/GameAnalytics.Build.cs

PhDittmann, 2018-06-27:
see #37

closed 2018-06-28 by Sanic