code-iai/ROSIntegration #39: Possibility to create new messages/services etc. outside of this plugin

PhDittmann, 2018-06-27:
Right now it is impossible to create new ROS messages / services / converter etc. outside of this plugin, because all base classes are private and not prepared to be available in a shared library. This means, this problem can't be solved by just moving the classes to the Public folder. At this ways it's quite hard to keep custom files seperated from this repository.

Hemofektik, 2018-06-27:
Thanks for the input. Currently, I'm about to implement a service in a seperate plugin and will do the necessary changes to the ROSIntegration Plugin that are necessary to achieve this.

Hemofektik, 2018-06-28:
Hey ho, I tried it and it worked out of the box as it is currently.
Here you can see how I did it:
https://github.com/Real-Synth/ROSControls/blob/master/Source/ROSControls/Private/ROSControlsGameInstance.cpp

One caveat is probably the Private Include paths that are relative to the other plugin
https://github.com/Real-Synth/ROSControls/blob/master/Source/ROSControls/ROSControls.Build.cs
But this could be fixed by putting those headers into public folders as you suggested.

Anything else that you encountered that kept you from creating a service/topic/messages converters in your own plugin?

Hemofektik, 2018-07-26:
Do you still have any issues or does it work for you now?

jteuber, 2018-11-12:
@Hemofektik I need to do this now too, but the files you linked are not on github anymore. Could you outline what you did in the cpp and build.cs files?

Hemofektik, 2018-11-12:
@jteuber Sorry, the repository was private. I changed it back to be public again. If anything remains unclear, let me know.

jteuber, 2018-11-13:
@Hemofektik Thank you! I got it to work now.
For anyone reading this later, I had to add the following in my projects Build.cs:

```
    string rosintegrationPath = Path.GetFullPath(Path.Combine(ModuleDirectory, "../../Plugins/ROSIntegration/Source/ROSIntegration/Private"));
    PrivateIncludePaths.AddRange(
        new string[] {
            rosintegrationPath,
            rosintegrationPath + "/rosbridge2cpp"
	});
```

This seems to be necessary if you are in a UProject. Everything else was indeed pretty straight forward.

closed 2018-11-21 by Sanic

PhDittmann, 2019-01-30:
The refactoring worked as easy as you said. Just added following to the Build.cs:

```
    string ROSIntegrationPath = Path.GetFullPath(Path.Combine(ModuleDirectory, "..", "..", "Plugins", "ROSIntegration", "Source", "ROSIntegration", "Private"));
    PrivateIncludePaths.AddRange(new string[] {
        rosintegrationPath,
        Path.Combine(rosintegrationPath, "rosbridge2cpp")
    });
```

and changed the DLL export macro in front of the Conversion classes from ROSINTEGRATION_API to *YOURPROJECTNAME*_API.