code-iai/ROSIntegration #42: Compile error when building a sample project using the plugin

carluciosk8, 2018-08-08:
Hello Guys

I'm trying this ROSIntegration plugin for first time and I'm having the following compile error:

error C2039: '_Ric': is not a member of 'UROSIntegrationGameInstance' 

I created a C++ project using the Advanced Vehicle Template and I have followed all the steps in the Usage section of the README with success. But when I moved to the next section "C++ Topic Publish Example" I did like the instructions: I added the code into the BeginPlay() of the Pawn class and I've got this error.

Thanks

Hemofektik, 2018-08-08:
Ah sorry, we missed to update the sample code.
Change "rosinst->_Ric" to "rosinst->ROSIntegrationCore" and it should compile

Sanic, 2018-08-08:
I just fixed this in the README