code-iai/ROSIntegration #48: Failed plugin build in UE4.20

peachthiefmedia, 2018-10-27:
The recent update to SensorMsgsImageConverter seems to be blocking the plugin from building now, at least in UE 4.20 error in build log is

> SensorMsgsImageConverter.h(37): error C2039: 'data': is not a member of 'ROSMessages::sensor_msgs::Image'
> Image.h(8): note: see declaration of 'ROSMessages::sensor_msgs::Image'

If you comment out line 37 it builds fine; don't think it is due to building on 4.20 looking at it quickly.

Hemofektik, 2018-10-27:
There is already a PR #47 to fix this issue.

peachthiefmedia, 2018-10-27:
Sorry I missed looking for it in a pull, I can close this or leave it here for people trying to build in 4.20 until the request is accepted?

Hemofektik, 2018-10-27:
No problem, as you already found out it should run fine with 4.20 with the fix. Feel free to leave the issue open until the PR is merged.

Sanic, 2018-11-21:
PR is merged -> Closing this issue

closed 2018-11-21 by Sanic