code-iai/ROSIntegration #52: Unreal Engine crashes when creating a new C++ classes inside of the plugin

Freya07, 2018-11-20:
Hi all, 
I'm working on Unreal Engine 4.19 installed on a Windows 10 machine, I managed to setup the plugin properly and to subscribe to an existing ROS node succesfully. 
Still I can't create new C++ classes inside of the ROSIntegration module because UE crashes everytime I tried to do so, it allows me just to create class inside the main directory of the Project. 
I had the same issuses when I tried, according to the example, to create a new C++ class based on "ROSIntegrationGameInstance" (Blueprint class worked fine). 
I attach you the callstack after UE crashes. 
Did anyone have the same problem? Do you know how could I solve it?
Thanks 
[callstack.txt](https://github.com/code-iai/ROSIntegration/files/2599762/callstack.txt)

Hemofektik, 2018-11-20:
You are right. It crashes when using the new C++ Class Wizzard for adding a new class to the Plugin. Looks like BeginDestroy is called twice here which seems to happen for the root object of the Topic during hotreload of the plugin.

I submitted a PR (#53) which solves this issue,

Sanic, 2018-11-21:
The mentioned PR has been merged into the repository.

If you still encounter this error with the new PR, please let us know in a new or reopened issue.

closed 2018-11-21 by Sanic

Freya07, 2018-11-21:
Thank you so much, now it works perfectly fine!