code-iai/ROSIntegration #59: Filling in missing standard message types

btwolfe, 2019-01-07:
I want to use the sensor_msgs/JointState message type and I see that it is not in the ROSIntegration repo. This is a pretty common message if you're doing anything with robots, so I'm surprised it's not there. Anyway, has this message been implemented by anybody to your knowledge? I started to implement it but thought it worth asking.

jteuber, 2019-01-08:
Oh, I actually implemented that message a while ago but haven't gotten to pushing it yet. I can do that later today if you haven't already implemented it yourself.

btwolfe, 2019-01-08:
Please let me know when you do. I'm running into some difficulty in getting the arrays out of a received message. Keep getting a "LogTemp: Error: Key msg.name not present in data"

jteuber, 2019-01-08:
Oh, I'm sorry. I implemented a custom message that's also called JointState and got confused. I haven't done the sensor_msgs-one yet. But maybe I can help you if you show me what you got so far.

btwolfe, 2019-01-08:
Well, I got it working. I had the `GetFStringTArrayFromBSON("msg.name", message->full_msg_bson_, keyFound, true);` code right but did not assign it to `p->name` so my ROS listener actor was throwing the error. Haven't done the outgoing message yet. Now just need to get urdf into UE4, parse it, and figure out how to translate those JointStates into a proper pose. I don't see any support in ROSIntegration for accessing the parameter server.

Sanic, 2019-01-11:
ROSIntegration doesn't support it out of the box, but atleast the communication protocol that we use (rosbridge) should allow you to do so http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality + https://github.com/RobotWebTools/roslibjs/blob/develop/src/core/Param.js

closed 2019-03-05 by Sanic