code-iai/ROSIntegration #60: tf2_msgs types

btwolfe, 2019-01-10:
The ROSIntegration landing page has a table showing the messages that have been implemented. I see that tf2_msgs/TFMessage is marked as complete for UE4 to ROS but not the other way around. Is that still the case or is there some issue that made implementing it the other direction particularly difficult?

btwolfe, 2019-01-10:
After some investigation, I see that the IncomingMessage process would require a loop over an unbounded array of geometry_msgs/TransformStamped messages, which is not implemented for receiving. That depends on geometry_msgs/Transform, which is also not implemented for receiving. I'm starting to see the challenge. The components of a transform (i.e., Vector3 and Quaternion) are implemented for receiving, so that's promising. Figure I can start building up from there. Maybe this is something you've already implemented and just haven't pushed yet? (fingers crossed)

btwolfe, 2019-01-16:
Shall I push changes to your repository that implements these types?

Sanic, 2019-01-18:
If you wanna share them, please open a Pull Request and include them there.

sparky002, 2020-02-14:
Any progress on this?

btwolfe, 2020-02-14:
Totally forgot about this. I just forked the repo, added my changes, and issued a pull request.

sparky002, 2020-02-14:
> Totally forgot about this. I just forked the repo, added my changes, and issued a pull request.

Awesome! Thanks for the quick response! Can't wait to try this out!

btwolfe, 2020-02-17:
I got an email saying the changes have been merged into master, so you should have access to them now.

closed 2020-02-17 by btwolfe

sparky002, 2020-02-17:
After the latest merge, the plugin does not compile. It might be related to the code you pushed. Tried on a new c++ ue4 project.

ActionlibMsgsGoalStatusArrayConverter.h:48:51: error: cannot initialize a parameter of type 'ROSMessages::std_msgs::Header *' with an rvalue of type 'const std_msgs::Header *'
                UStdMsgsHeaderConverter::_bson_append_header(b, &(g->header));
                                                                ^~~~~~~~~~~~

GeometryMsgsPoseStampedConverter.h:46:51: error: cannot initialize a parameter of type 'ROSMessages::std_msgs::Header *' with an rvalue of type 'const std_msgs::Header *'
                UStdMsgsHeaderConverter::_bson_append_header(b, &(ps->header));

GeometryMsgsTransformStampedConverter.h:6:10: fatal error: 'TransformStamped.h' file not found
#include "TransformStamped.h"

sparky002, 2020-02-18:
A new pull request was made to fix the BSON errors. To fix the transformstamped error, you just have to add "geometry_msgs/" before "TransformStamped.h" in the include statement in the GeometryMsgsTransformStampedConverter.h file.

btwolfe, 2020-02-18:
Odd, I used this code without issue a year ago so didn't think there would be any issues. I don't use this code anymore but obviously I should have tested again before my push. Thanks for the correction.