code-iai/ROSIntegration #62: Rosbridge connected but no published topics visible

phil333, 2019-01-18:
when launching the rosbridge_server: roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True

I get the confirmation that it connected:
[INFO] [1547805756.197721]: Rosbridge TCP server started on port 9090
[INFO] [1547807016.693514]: [Client 0] connected. 1 client total.

The rostopic list does not show anything
/client_count
/rosout
/rosout_agg

The rosbridge_server lists the following versions:  
* /rosdistro: melodic
* /rosversion: 1.14.3
Running on ubuntu 18.04

I already tried this:
https://github.com/RobotWebTools/rosbridge_suite/issues/353

Also running Unreal in a separate Windows machine, resulted in the same problem

The C++ code in unreal is a copy paste of the example in the readme, copied into:
void AMyActor::BeginPlay()

I had the test example working before, but i have run some updates in the meantime. So it is possible that an updated dependency has caused this.

I also tried using the blueprint setup, but nothing is published. I also tried the subscribe method, but for some reason, my subscribe blueprint does not give me the same options displayed in the readme file.

phil333, 2019-01-21:
I have narrowed manage to narrow down the problem to the rosbridge_server on this machine.

When trying to run everything on another machine (also ubuntu18 and ROS-melodic), everything worked. I then tested both rosbridge_server and unreal with either machine over tcp/ip. rosbridge_server still reports a connection with Unreal running on a separate machine, but there is no visible topic published on the rostopic list.

Any suggestions on how to debug this or maybe some dependency version suggestions?

phil333, 2019-01-21:
I tried subscribing, without any result. Still need to try this with the second machine setup.
The topic declaration works flawlessly when connecting this Unreal project to a rosbridge_server that runs on a separate machine.

phil333, 2019-01-21:
Found the issue. I recently installed Anaconda for a separate project. This seems to have changed my python environment. While i dont know which specific library caused the issue, removing anaconda from the .bashrc file solved the problem.

closed 2019-01-21 by phil333

MahyarBayran, 2021-04-21:
@phil333 so which python path worked for you? Right now, mine is using the one integrated in ros noetic package.

[image]

phil333, 2021-04-21:
@MahyarBayran as mentioned, my issue was just to remove the added paths from anaconda which ruined things in the first place. I was also using melodic, so nothing with python3. I have since moved on to using Airsim instead. Its not perfect either, but easier to set up. I needed a stereo camera configuration, and i didn't know enough about unreal to modify the existing code samples.