code-iai/ROSIntegration #63: LZ4 support failed

phil333, 2019-01-18:
When launching the rosbridge server as follows, i get an error message:
`roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True`

The error message is:
`Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
`

The full console readout:

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_tcp/authenticate: False
 * /rosbridge_tcp/bson_only_mode: True
 * /rosbridge_tcp/delay_between_messages: 0
 * /rosbridge_tcp/fragment_timeout: 600
 * /rosbridge_tcp/host: 
 * /rosbridge_tcp/incoming_buffer: 65536
 * /rosbridge_tcp/max_message_size: None
 * /rosbridge_tcp/params_glob: [*]
 * /rosbridge_tcp/port: 9090
 * /rosbridge_tcp/retry_startup_delay: 5
 * /rosbridge_tcp/services_glob: [*]
 * /rosbridge_tcp/socket_timeout: 10
 * /rosbridge_tcp/topics_glob: [*]
 * /rosbridge_tcp/unregister_timeout: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_tcp (rosbridge_server/rosbridge_tcp)

ROS_MASTER_URI=http://localhost:11311

process[rosbridge_tcp-1]: started with pid [12234]
process[rosapi-2]: started with pid [12235]
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
registered capabilities (classes):
 - <class 'rosbridge_library.capabilities.call_service.CallService'>
 - <class 'rosbridge_library.capabilities.advertise.Advertise'>
 - <class 'rosbridge_library.capabilities.publish.Publish'>
 - <class 'rosbridge_library.capabilities.subscribe.Subscribe'>
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - <class 'rosbridge_library.capabilities.advertise_service.AdvertiseService'>
 - <class 'rosbridge_library.capabilities.service_response.ServiceResponse'>
 - <class 'rosbridge_library.capabilities.unadvertise_service.UnadvertiseService'>
trying to start rosbridge TCP server..

Sanic, 2019-01-18:
Hi! This looks related to rosbridge and not ROSIntegration. so i would suggest to open a issue there: https://github.com/RobotWebTools/rosbridge_suite/issues

phil333, 2019-01-18:
my bad, that's where i wanted to post it. I apologize for the spam

closed 2019-01-18 by phil333