code-iai/ROSIntegration #65: accessing class instance variables in subscriber callback

btwolfe, 2019-01-23:
Example code for a subscriber demonstrates accessing content of a message and displaying it without storing it inside of a lambda function. For practical applications, you really want to capture message content and store it with an instance of the listener class. It's not clear to me how to change that lambda function into a class instance method call or pass an instance of the class to the lambda function. Suggestions?

Sanic, 2019-01-24:
iirc i've used https://de.cppreference.com/w/cpp/utility/functional/bind for that.

btwolfe, 2019-01-24:
Yes, that's what I eventually settled on, just thought you might have a different approach. Consider modifying your example code to use std::bind instead of the lambda function.

closed 2019-01-24 by btwolfe

FiaTNessa, 2019-06-17:
Could you please give any hints on how to do this? I mean, changing the subscribe for a bind function? Or is that any way we can access the callback content?
Thanks in advance.

Sanic, 2019-06-18:
This is what i used in UMyClass::BeginPlay to bind the Callback to a member function

 _listener->Subscribe(std::bind(&UMyClass::MessageCallback, this, std::placeholders::_1));

FiaTNessa, 2019-06-18:
Tried to create the Callback from my actor class, but Subscribe() forces it to be a: std::function<void(TSharedPtr<FROSBaseMsg>)>

So I decided to create a global variable and access it in the lambda callback function.
Thank you for you response.

FiaTNessa, 2019-06-18:
I tried to create the Callback from my actor class, but Subscribe() forces it to be a: std::function<void(TSharedPtr)>

How should I create the callback then?

Sanic, 2019-06-19:
Hi.
std::bind can be used for std::function parameters. 
See an example that i just built:
http://cpp.sh/2krlr

Please note that the signature of your callback function must comply with the template of the std::function, meaning it must return void and accept a TSharedPtr

FiaTNessa, 2019-06-19:
Hi there,

Just want to say thank you for your help, i can now access data outside from the callback. However, if I don't check this: receivedmsg.IsValid(), Unreal crashes

But the problem is that I can only receive a few messages, and after some time the connection goes down, the message "Found CB in UnregisterTopicCallback. Deleting it" appears, and after a few seconds the connection goes up again, I receive a few more messages... It is not continuous.

Could the reason be the higher topic publishing frequency? Is there a buffer parameter configuration for received messages?

Thanks.