code-iai/ROSIntegration #71: Matching Motion in Gazebo and UE4

smithan7, 2019-03-04:
I am working on a project where I am simulating the cameras in UE4 and some of the sensors in Gazebo, e.g., IMU/GPS/Magentometer using the Hector Package, and therefore need the quadcopter to move in Gazebo identically to how it does in UE4. I think I can 'hack' a solution by controlling the quad in Gazebo and subscribing to the Gazebo published odometry topic in UE4 then 'teleporting' the quad to the published odom topic. However, I am wondering if there is a cleaner or suggested way to accomplish this.

Thanks

madratman, 2019-03-04:
https://bitbucket.org/osrf/gzue4bridge/wiki/Home might be something to look at.
I'm interested in this as well

Sanic, 2019-03-05:
Hi!

I would look into the tf2 Package (http://wiki.ros.org/tf2) which is also supported by this Plugin.
You could then try to publish the pose of your quad in UE4 to the TF topic, read that pose from gazebo and then set up the quad there accordingly.

closed 2020-02-17 by Sanic