code-iai/ROSIntegration #73: Convert between coordinate systems in messages?

jteuber, 2019-03-11:
Hi there!
While working with a visual SLAM I realized that unreal and ROS use two different coordinate systems that, for simple use-cases don't really interfere, but which might be a problem later on.
According to the ROS docs (http://www.ros.org/reps/rep-0103.html), it uses a right-handed coordinate system with x forward, y left and z up. Unreal on the other hand uses a left-handed one with x forward, y right, z up. 
So only the Y-coordinate changes sign, which is fine as long as everything stays in that coordinate system. ROS will just operate in a mirrored version of the unreal space. Unfortunately, when visual information is used, that becomes a problem.
So my question is, should we maybe change all messages that have to do with coordinate systems (mostly geometry_msgs/Vector3 and Quaternion) to automatically convert between the coordinate systems or should the user do that themselves?

Sanic, 2019-03-12:
Hi!

That's a very interesting point that also hit me when building ROSIntegrationVision. There, i directly included the conversion while publishing tf. In that case, it's necessary to convert the coordinates between the tf tree has to be constructed properly.
For general Vectors and Quaternions i would be a bit more defensive, as people might use Vectors for all kinds of stuff, and not always to explicity describe an (absolute) location in the world. I'm also with you on your con "might lead to confusion about what the messages do in the background".

I would propose that one should be able to supply a flag for every instantiation of a message/service which enables the conversion everytime a message is processed (given, the message/service handles coordinates).
What do you (and others looking here) think of that?

jteuber, 2019-03-13:
A flag for the messages/services sounds like a good idea, then it can also be backward compatible. The problem might be, that that flag needs to be implemented in at least all messages that use Vector3 and Quaternion and be passed down. Seems to me like a lot has to be changed for that to work. Also, how does the topic know into what coordinate system it should interpret the incoming messages?

Maybe another option would be to have a flag in the topics, passed down to the converters which make sure that they convert incoming messages from the ROS-side to Unreal coordinates and vice versa. That would still need a lot of changes though.