code-iai/ROSIntegration #77: ROSIntegration.uplugin is not being compiled

kamilhassaan, 2019-03-20:
I have been trying to connect ROS with **Unreal 4.20.3** but all my tries go in vain. The plugins folder was not present by default when the project was created. I created this manually in the directory and created another folder inside plugins named **ROSIntegration**. I pasted **ROSIntegration.uplugin** inside **ROSIntegration** folder. After that when I start the project, i get the following messages:
_**The following modules re missing or built with a different engine version**_
_**ROSIntegration**_
_**Would you like to rebuild now**_

[image]

I click yes and then this message appears:
_**MyProject2 could not be compiled. Try rebuilding from the source manually**_
[image]

I click okay and unreal closes. How can this problem be fixed?

samkys, 2019-03-20:
Did you only paste the ROSIntegration.uplugin into the ROSIntegration directory?

Is there any other files in the Plugins/ROSIntegration directory?

kamilhassaan, 2019-03-21:
Yes. I only pasted ROSIntegration.uplugin into the ROSIntegration directory.

No. There are no other file(s) in the  Plugins/ROSIntegration directory.

These are the screenshots of the directory:

[image]

[image]

[image]

Sanic, 2019-03-21:
Hi

1) Having no Plugins/ folder a new project is normal. You usually have to create them manually.
2) You have to put the whole repository content into the Plugins/ folder. Not only the uproject file...:
From the readme:
_Add this repository to your Plugins/ Folder in your Unreal project (**copy the folder in** so your structure looks like MyUnrealProject/Plugins/ROSIntegration/ROSIntegration.uplugin_

samkys, 2019-03-21:
As stated above already, just make sure everything that you cloned or downloaded from github.com/code-iai/ROSIntegration is in that directory as well as the ROSIntegration.uplugin. It should work fine after that.

closed 2019-03-21 by Sanic

kamilhassaan, 2019-03-25:
Thank you! That worked