code-iai/ROSIntegration #79: UGeometryMsgsTransformStampedConverter: Needs work

samkys, 2019-03-30:
In the readme it states geometry_msgs/TransformStamped are supported from UE4->ROS 
& ROS->UE4

ROS->UE4 doesn't appear to be supported even though the readme says it is.

See: line 15 of UGeometryMsgsTransformStampedConverter.cpp 
bool UGeometryMsgsTransformStampedConverter::ConvertIncomingMessage(const ROSBridgePublishMsg* message, TSharedPtr<FROSBaseMsg> &BaseMsg)
{
	UE_LOG(LogROS, Warning, TEXT("ROSIntegration: TransformStamped receiving not implemented yet"));
	return false;
}

Next comment: UE4->ROS of geometry_msgs/TransformStamped are stated to work in the Readme. Has this been tested? I have implemented it but it just crashes Unreal. (I am trying to find out why...)

ReeRichard, 2019-04-12:
could you please share the code you tried?

samkys, 2019-05-20:
Sorry for the delay in responding, I have been busy. Anyhow, I have created a message function and pass in a transform. See code below.

```// TF Stamped message Publisher
void ARosActor::tfStampedMsg(FTransform transform, FString frame_id, int32 msg_seq) 
{

  // Create the new topic
  UTopic* tf_stamped_topic_ = NewObject<UTopic>(UTopic::StaticClass());
  
  // Get the ROS game instance
  UROSIntegrationGameInstance* rosinst = Cast<UROSIntegrationGameInstance>(GetGameInstance());
  
  // Initialize the Topic
  tf_stamped_topic_->Init(rosinst->ROSIntegrationCore, TEXT("/tf_stamped"), TEXT("geometry_msgs/TransformStamped"), 1);
  
  // Advertise our publisher
  tf_stamped_topic_->Advertise();
    
  // Create an instance of our TF Stamped message
  TSharedPtr<ROSMessages::geometry_msgs::TransformStamped>
    tfStampMessage(new ROSMessages::geometry_msgs::TransformStamped());
  
  //-------------------Message header creation and population-----------//
  tfStampMessage->header = ROSMessages::std_msgs::Header(msg_seq, FROSTime::Now(), frame_id);
  tfStampMessage->child_frame_id = FString("ref_transform");
  FVector tf = transform.GetTranslation();
  tfStampMessage->transform.translation = ROSMessages::geometry_msgs::Vector3(tf.X,tf.Y,tf.Z);
  tfStampMessage->transform.rotation = ROSMessages::geometry_msgs::Quaternion(transform.GetRotation());
  
  // Publish.....
  tf_stamped_topic_->Publish(tfStampMessage);
}```

Sanic, 2020-02-17:
I just merged a PR from a contributor who thankfully implemented tf-related messaging.
I'm therefore closing this issue. If you think this todo persists, please update your version of ROSIntegration, re-check and re-open the issue if it's still there.

closed 2020-02-17 by Sanic