code-iai/ROSIntegration #90: Where to add publisher and subscriber code?

nabeelyousfi, 2019-07-04:
Hi.

I am new to Unreal Engine. I have used ROS with Gazebo and VREP but not with UE4. I added the ROSIntegration plugin and its successfully added. But now it says to 

> create a C++ Actor and let it publish a message once at the BeginPlay Event. Add the following code into the BeginPlay() method of any actor that is put into to your world to see if the connection to ROS works

I don't know where to start, where to add the code, and where can I see if my publisher/subscriber is working? I am stuck with this since 3 weeks. I would really appreciate if someone would help me with this.

mp3242, 2019-07-22:
I'd like clarification on this as well. No matter how I do it, I get errors while importing the header files from the Public folder. But I may just be doing it wrong, so here's what I'm doing.

I followed the tutorial for the most part (enabling the plugin was a little different than the readme). But after enabling the plugin, creating a blueprint based on ROSIntegrationGameInstance, and editing the Project Settings, I went to the content browser and clicked Add New > New C++ Class > Actor, and put it into ROSIntegration (Developer) (Public), and created the class. After the code compiles I can double click on C++ Classes > ControlCenter > Public > MyActor and add the subscriber code from the readme into the AMyActor::BeginPlay function, under the Super::BeginPlay. When I add the include lines though, the Image.h file is unrecognized. No matter what I do I can't import files from the Public folder, and it doesn't show up in the content browser either (along with rosbridge2cpp curiously)

sparky002, 2019-10-07:
More clarification on where and how to place the code would be useful in the documentation. Things like creating the actor to ROSIntegration(Developer)(Public) would be useful. This is something other users have to figure out from experimentation, or look through these issues. Also, would be useful to know that you have to re-compile headers after any ros c++ code change in your project for the change to occur when playing back the unreal sim or document a method to prevent the need to do this. A complete basic UE4 ROS example project both c++ and blueprints would also be incredibly useful for people that aren't sufficient in UE4 yet.

closed 2019-11-22 by nabeelyousfi