code-iai/ROSIntegration #93: Publishing (of simple string) stops after ~30seconds of frequent publishing

p-buddy, 2019-08-29:
Hi!

I'm experimenting with setting up a string publisher (a very simple one exactly as described in the README tutorial). To experiment with how frequently I could publish, I moved the publish call into the Tick() function.

When I do so, I observe the message publishing at ~3Hz (as confirmed by doing a 'rostopic hz' on the topic). After about 30 - 60 seconds, no new messages will be published to the topic. 

Occasionally, I'll see the output of the 'rostopic hz' call balloon up (from ~3Hz to ~30Hz) before the messages stop publishing. 

I've also seen an error thrown in Unreal that causes the editor to crash, but unfortunately didn't save that output. If I can get it to happen again, I'll attach it here. 

Here are my specs:
Unreal: 4.22.3 (I see it wasn't tested on this version -- would/could that be an issue)
OS: Ubuntu 16.04

Are their limitations to publishing size and frequency?

samkys, 2019-10-28:
Assuming the time is random like you said this sounds like a garbage collector issue. It's on the UE4 side of things, I believe what happens is that UE4 thinks you are no longer using the pointer so it tries to delete it. For a quick test to see if that is the problem adjust the time like in the image attached.
(To get there, go to `edit->project settings` then `engine->garbage collection` change the piece that's circled to something way less and see what that does. Then change it to something way longer and see what that does.

If that solves the problem what you need to do is something like this:

```  // Topics for subscribers!
  // These need to be property'd so that they don't get garbage collected
  // https://github.com/code-iai/ROSIntegration/issues/32
  UPROPERTY()
  UTopic* topic_;
```

[image]

samkys, 2019-12-04:
@p-buddy Any news? Would be great to close this if that was the problem.

rkvtmw, 2020-03-22:
I found this issue after experiencing the same symptoms. Somewhere between 10-30s of time running Unreal, I would get an access violation in `Topic.h`. Adding `UPROPERTY()` to my member declaration of UTopic seemed to fix it.