code-iai/ROSIntegration #96: Support for Unreal 4.23

ashepele, 2019-10-21:
ROSIntegration Team,

I've been using the ROSIntegration plugin successfully in 4.22, but have started seeing an issue (please see screenshot below) related to garbage collection in 4.23. The issue can be easily reproduced with a blank project, so I'm fairly certain the issue is an incompatibility between the ROSIntegration plugin and Unreal 4.23.

What are the team's plans regarding support for Unreal 4.23? 

**Steps to reproduce crash:**
- Create blank project in Unreal 4.23
- Add ROSIntegration as a plugin to the blank project
- Open the new project in the Unreal Editor
- Start and stop simulation
- Close the editor (crash occurs here)

[image]

Sanic, 2019-10-22:
Hi!
I just merged https://github.com/code-iai/ROSIntegration/pull/95
Can you please check if the problem still persists?

ashepele, 2019-10-23:
Hi Sanic,

The new updates fixed it. Thank you!

closed 2019-10-23 by ashepele