code-iai/ROSIntegration #101: Compatibility with Unreal 4.24

ashepele, 2019-12-10:
ROSIntegration Team,

Epic just released 4.24 and I tried running ROSIntegration with it. I'm running into an issue with the AddUObject command. Could you assist with this one, please?

Error log is below.

[image]

Below is the test version of the error, in case there is a need to copy/paste any of it. 

`ROSIntegrationGameInstance.cpp:86:38: error: no matching member function for call to 'AddUObject'
                        FWorldDelegates::OnWorldTickStart.AddUObject(this, &UROSIntegrationGameInstance::OnWorldTickStart);
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/ashepele/UnrealEngine4.24/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:903:25: note: candidate function [with UserClass = UROSIntegrationGameInstance, VarTypes = <>] not viable: no known conversion from 'void (UROSIntegrationGameInstance::*)(ELevelTick, float)' to 'typename TMemFunPtrType<false, UROSIntegrationGameInstance, void (UWorld *, ELevelTick, float)>::Type' (aka 'void (UROSIntegrationGameInstance::*)(UWorld *, ELevelTick, float)') for 2nd argument
        inline FDelegateHandle AddUObject(UserClass* InUserObject, typename TMemFunPtrType<false, UserClass, void (ParamTypes..., VarTypes...)>::Type InFunc, VarTypes... Vars)
                               ^
/home/ashepele/UnrealEngine4.24/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:910:25: note: candidate function [with UserClass = UROSIntegrationGameInstance, VarTypes = <>] not viable: no known conversion from 'void (UROSIntegrationGameInstance::*)(ELevelTick, float)' to 'typename TMemFunPtrType<true, UROSIntegrationGameInstance, void (UWorld *, ELevelTick, float)>::Type' (aka 'void (UROSIntegrationGameInstance::*)(UWorld *, ELevelTick, float) const') for 2nd argument
        inline FDelegateHandle AddUObject(UserClass* InUserObject, typename TMemFunPtrType<true, UserClass, void (ParamTypes..., VarTypes...)>::Type InFunc, VarTypes... Vars)
`

Sanic, 2019-12-11:
Hi!
#100 has just been merged. Can you please try to compile with the most recent version of ROSIntegration?

ashepele, 2019-12-11:
@Sanic Thank you and thank you to @Tim-Fronsee for the very quick response time on this! I tested the changes that they build in 4.24. We will do some additional testing in this new version of Unreal and reach out if there are any issues. Thanks again!

Sanic, 2019-12-17:
@ashepele Is everything look good so far?

ashepele, 2019-12-17:
So far so good, thanks. I will go ahead and close the issue

closed 2019-12-17 by ashepele