code-iai/ROSIntegration #102: ROS2 Support?

russkel, 2019-12-14:
Hello,

How much work is required for this project to work with ROS2 as well?

Sanic, 2019-12-17:
I guess it would mainly necessary to either write a CPP client for https://github.com/RobotWebTools/ros2-web-bridge
or check if ROS2 could be directly integrated as a library on Linux, Windows and maybe even MacOS. I heard that this might be easier with ROS2 than with ROS1.

russkel, 2019-12-17:
Oh yeah, never thought about using the web bridge. Looks like it's JSON based: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md
That would probably be fairly straightforward.

Sanic, 2019-12-17:
You raise an interesting point.
ROSIntegration uses the BSON mode of rosbridge(1) to get faster transmission rates.
Converting everything to JSON is rather slow, which is why i mainly integrated BSON into this Plugin.
By quickly looking over the rosbridge2 code, i can't see a bson mode yet.
Even though it shouldn't be too hard to add.

russkel, 2019-12-17:
That seems like a good place to start. I'll start an issue there and see where it goes.

russkel, 2020-11-13:
So the new project SOSS looks like an interesting solution to this problem: https://soss.docs.eprosima.com/en/latest/

tsender, 2021-04-21:
You can also use the ROS1 Bridge: https://github.com/ros2/ros1_bridge

I have successfully been able to communicate between the ROSIntegration plugin and ROS2 with this approach. To keep matters simple, all you really need to is make sure whatever messages you are using have a definition in both ROS1 and ROS2. The ros1_bridge also has documentation on how to build the bridge with custom messages.

thejeshk, 2021-07-29:
> You can also use the ROS1 Bridge: https://github.com/ros2/ros1_bridge
>
> I have successfully been able to communicate between the ROSIntegration plugin and ROS2 with this approach. To keep matters [...]

Hello @tsender 

I am also planning to use ROS2 with Unreal Engine. Do you have concrete documentation how to integrate them?

Thank you 
TK

tsender, 2021-07-31:
Hi @thejeshk 

I do not have concrete documentation for linking ROS2 to UE4. The addition of ROS2 mostly consists of following the ros1_bridge documentation, which isn't too hard to follow. But I can provide some helpful tips/suggestions and other useful information. Assuming your main ROS version is ROS2:

1. If you use custom ROS messages or services, then you should
   - Create a ROS1 repo for the sole purpose of containing the ROS1 definition of the messages/services
   - Create the definition for the messages/services in your ROS2 workspace and in UE4 (obviously)
   - Follow the steps in the ros1_bridge documentation to build the ros1_bridge with your custom messages/services

2. If you only work with ROS topics, then you should not run into any issues using the ros1_bridge between ROS2 <--> UE4.

3. If you require ROS services/clients, then things might become more involved.
   - If you have a UE4 client and a ROS2 service, then you (generally) will not have any problems between ROS2 <-->UE4.
      - The only problem you may have is if your ROS2 service requires >=5 seconds to process the request, then after the 5s grace period the ROSIntegration plugin will immediately give UE4 an empty response (I believe this is because the plugin constantly checks its health every 5s) and will disregard whatever the ROS2 service responds. If you know your ROS2 service response takes >=5s to process, or if you need to use a ROS-style action, then you will need to split it up into 2 service calls, in which sending the request is its own service call (UE4 --> ROS2) and sending the response is its own service call (ROS2 --> UE4). Unfortunately, using these 2 service calls now forces you to follow the steps in the next bullet.
   - If you have a ROS2 client and a UE4 service, then for some unknown reason the ros1_bridge will _not_ create a 2-to-1 bridge  to send the service request from ROS2 to UE4. The only way around this (that I know of) is to manually create a ROS1 middleman. Creating this middleman forces the ros1_bridge to create the needed 2-to-1 bridge. In this middleman, you will create a ROS1 node with both a service and a client. The process is as follows: The ROS2 client sends a service request to the ROS1 service, the ROS1 service then uses this info and uses the ROS1 client to create a service request to the UE4 service, then UE4 receives the data and processes it and send its response to the ROS1 client which send it back up the chain to the ROS2 client. Visually, the process looks like this:
     - Request pipeline (left to right): ROS2 client --> ROS1 service --> ROS1 client --> UE4 service
     - Response pipeline (right to left): ROS2 client <-- ROS1 service <-- ROS1 client <-- UE4 service

As you can clearly see, using a ROS service/client can potentially add a lot of additional steps just to make this connection work. It is totally doable, but can be a bit tedious. If you can get away with just ROS topics, then do that. Only use the services if you need to ensure data is sent/received (which I had to do for my project).

These are probably the most important tips I can provide for linking ROS2 <--> UE4. I hope you and anyone else pursuing this route find this post helpful. Please let me know if you have any other questions.

russkel, 2021-07-31:
I think people overestimate how difficult it is to port things across to ROS2. I don't have a use case for this bridging yet but once I do I'll do the work. But people shouldn't shy away from ripping out the ROS1 stuff and getting it to work with ROS2 if they need it.

GaoGeolone, 2021-09-16:
@Sanic @russkel @tsender @thejeshk @mareikep 
Hey Guys, I made it work with ROS2. https://github.com/RobotWebTools/rosbridge_suite/issues/639#issuecomment-920504652
[image]

Sanic, 2021-09-16:
Nice job!
I'm crossing my fingers that this change gets into the official rosbridge_suite  👍

tsender, 2022-02-16:
@GaoGeolone, it seems that the current rosbridge_suite got rid of the TCP socket and only has the web socket to communicate with the ros2 version. Did you, or any one else, have intentions of modifying this plugin to work with a web socket? I see that UE4 seems to have a module for web sockets, but I'm not sure if it meets the needs to work with this plugin.

russkel, 2022-02-16:
As an update to this issue, I have forked https://github.com/rapyuta-robotics/rclUE, available here https://github.com/Greenroom-Robotics/rclUE

It's under heavy development for a project.

tsender, 2022-03-09:
I actually recently managed to add a websocket client to this plugin. You can checkout my fork here: https://github.com/tsender/ROSIntegration/tree/ros2

Unfortunately, there are a few issues with the rosbridge_suite websocket server that need to be addressed, which I mentioned in the README. When these issues are addressed and I have done more testing with it, I will make a PR to this repo. But anyone is welcome to test out my code in the meantime. I still need to test subscribing to messages.