code-iai/ROSIntegration #103: How to install rosbridge-suite on windows?

nitin-ppnp, 2019-12-16:
Sorry to ask this here but I am interested in ROSIntegration and don't know how to install rosbridge-suite on windows. I tried `choco` but it cannot find this package in the default `roswin` source.

Sanic, 2019-12-17:
Hi!
I haven't tested this on my own. My usual workflow was that i have a Virtual Machine (for example in VirtualBox) that runs Ubuntu with ROS. 
If anybody has ever successfully used ROS/rosbridge on Windows, then reporting here would be highly appreciated.

nitin-ppnp, 2019-12-17:
I am a little confused. The first line of the main readme says that the Windows platform is supported. How can I run **ROSIntegration** on Windows if I can't have the dependency (i.e. **rosbridge**) installed?

Btw, ROS is available for windows and the detailed instructions are given here https://wiki.ros.org/Installation/Windows

Sanic, 2019-12-17:
Windows is supported in the sense that the Plugin can be run on the Windows version of UE4. This is not necessarily the case with all possible solutions. Imagine that the interprocess communication between ROS and UE4 is written in a platform depenend way...

In this plugin, this is abstracted by using ROSBridge where client libraries are available for all major platforms. This plugin was also written when ROS on windows wasn't a thing yet and our group is mostly using ROS on Linux.

nitin-ppnp, 2019-12-17:
Thanks for taking the time to answer. People are using Windows subsystem Linux for ros on windows. I will try to run rosbridge there and unreal with ROSIntegration on windows. I hope it works out.

nitin-ppnp, 2019-12-18:
I am able to run rosbridge on WSL and receive messages on windows. Thanks for your help.

closed 2019-12-18 by nitin-ppnp

wetoo-cando, 2023-04-28:
> I am able to run rosbridge on WSL and receive messages on windows. Thanks for your help.

@nitin-ppnp did you have to do anything special to receive messages on windows? I can reach the rosmaster inside WSL2, can even subscribe to my topic, but no data comes through.