code-iai/ROSIntegration #106: how to use " C++ topic Publish Example"  ?

Take-23, 2020-01-22:
Could anyone let me know how to use " C++ topic Publish Example"  ? 
I did the procedures below but many errors occurred when I built my UE4 project.

1. Created a template C++project to confirm how "ROSintegration" plugin works.
2. Activated "ROSintegration" plugin in the project.
3. Created a new C++ Actor class.
4. Added the code written in " C++ topic Publish Example"  to the BeginPlay() method in the class after modifying some '#'include paths written in  " C++ topic Publish Example" .
5. Then built the UE 4 project, but many errors occurred.

Below are the error messages I got when I tried " C++ topic Publish Example" 
'''
Error	C2079	'FROSBaseMsg' uses undefined class 'ROSINTEGRATION_API'	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	5	
Error	C2059	syntax error: 'public'	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	7	
Error	C2588	'::~FROSBaseMsg': illegal global destructor	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	9	
Error	C4430	missing type specifier - int assumed. Note: C++ does not support default-int	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	9	
Error	C2365	'FROSBaseMsg': redefinition; previous definition was 'data variable'	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	9	
Error	C2059	syntax error: '}'	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	12	
Error	C2143	syntax error: missing ';' before '}'	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\Public\ROSBaseMsg.h	12	
Error	C1083	Cannot open include file: 'ROSIntegrationCore.generated.h': No such file or directory	MyProject3	c:\users\admin\documents\unreal projects\myproject3\plugins\rosintegration\source\rosintegration\classes\ROSIntegrationCore.h	7	
Error	MSB3073	The command ""C:\Program Files\Epic Games\UE_4.23\Engine\Build\BatchFiles\Rebuild.bat" MyProject3Editor Win64 Development -Project="C:\Users\admin\Documents\Unreal Projects\MyProject3\MyProject3.uproject" -WaitMutex -FromMsBuild" exited with code -1.	MyProject3	C:\Program Files (x86)\Microsoft Visual Studio\2017\WDExpress\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	49
 ''''

Environment：
Unreal Engine ：4.23 ( I would like to use Unreal Engine with this version )
ROSintegration ：commit id 028c2e1

samkys, 2020-03-22:
Can you post your .cpp and .h files for your actor? It would help to debug.

Take-23, 2020-03-25:
Thank you for your reply.
Also, sorry for not updating anything about this issue.

I have already solved this issue.

The reason why this issue occurred was just that  I forgot the setting in ".cs".
Also, one thing I have done aside from ".cs " to solve this issue is to change "std_msgs/String.h"" into "std_msgs/StdMsgsString.h" in the source code "C++ Topic Publish Example" 
Note: 
There is no ".h" named "String.h" in the folder now, then I changed it to the ".h" above.

Then everything is working well now.
Thank you so much.

Sanic, 2020-03-25:
Thanks for reporting.

closed 2020-03-25 by Sanic