code-iai/ROSIntegration #112: Examples of ROS custom messages?

sparky002, 2020-02-26:
Has anyone integrated a custom message with this plugin yet? There is a description on how to set it up but some confusion around it which an integrated example would help.

For example, a message made up of strings, bools, and pose stamps or any combination of standard ros messages types that are already supported by this plugin.

Thanks

sparky002, 2020-02-27:
I'm able to expose a custom message type into blueprints, but run into the issue of subscribing to the custom init topic. 
Rosbridge (in terminal) is throwing out an error while UE4 seems fine with it. :

 " [id: subscribe:name_of_topic:12] subscribe: unable to import msg class custom_msg_name from package std_msgs. Caused by 'module' object has no attribute 'custom_msg_name' "       

I put my custom message inside the folder std_msgs.
I developed a converter for the custom message as well.

samkys, 2020-03-22:
I haven't personally tried custom messages, just built some of the standard ones for the plugin. Did you source the workspace in ROS where your custom message is defined before launching the ROSBridge in the same terminal instance?