code-iai/ROSIntegration #113: Unreal Engine Crash after Stop Game

DominikAUT, 2020-03-04:
Hy,
I have created two blueprints with which I want to read data from ROS. The reading works so far, but when I quit the game the Unreal Engine crash. It often works once and it only comes to a crash the second time. Sometimes an error is reported, sometimes not. The problem occurs with both string and float messages. Does anyone have any idea what it could be?

[image]

[image]

[image]

samkys, 2020-03-22:
Are you running the as PIE (Play In Editor)?

And how do you exit? I have experienced similar results due to shutting down ROS before I exit the "game" play.

Try hitting esc in UE4 to exit the game instance, then close out your ROS publishers. See if that helps and report back.

Roman-Malinowski, 2020-04-24:
Hi,

I have the same issue when using a C++ Actor listening to a classic publisher on my machine (not publishing from UE4).
UE4 does not crash when I listen to a String type of ROS message, but it keeps crashing when listening to a Float32MultiArray type of message. I am exiting the simulation using the UE4 Stop button (same as ESC key) while the Publisher is continuing to publish messages, and then it crashes. For now, it seems that Unsubscribing from the topic when ending the simulation prevent UE4 from crashing.

I don't know if the problem comes from here, but I'll share my code in case anyone has the same issue at some point.

MyActor.h:
```
public:
	virtual void EndPlay(const EEndPlayReason::Type EndPlayReason) override;
```
MyActor.cpp:
```
void AMyActor::EndPlay(const EEndPlayReason::Type EndPlayReason)
{
	ExampleTopic->Unsubscribe();
	UE_LOG(LogTemp, Warning, TEXT("Called Unsubscribe()"));
	Super::EndPlay(EndPlayReason);

}
```

PetterVMC, 2022-06-24:
Had a similar problem where it often crashed during shutdown if I was subscribing to high-frequency topics.

It appeared that if a message arrived during shutdown, then there was a chance that it triggered a subscription callback during a sensitive/invalid state (don't remember the details, will have to revit this again). I solved it by adding the following code to [UROSIntegrationGameInstance::Shutdown()](https://github.com/code-iai/ROSIntegration/blob/master/Source/ROSIntegration/Private/ROSIntegrationGameInstance.cpp), just before the line where is calls MarkAllROSObjectsAsDisconnected(). This code makes sure all topics are unsubscribed, so that no subscription callbacks may be called during the rest of the shutdown procedure.

```
	for (TObjectIterator<UTopic> It; It; ++It)
	{
		UTopic* Topic = *It;
		Topic->Unsubscribe(); // Necessary to avoid potential crash if messages arrive during shutdown.
		Topic->Unadvertise(); // Maybe not necessary? Added it just in case..
		Topic->MarkAsDisconnected();
	}
```

The fact that it happened often for high-frequency messages seems logical, as there is a much greater chance for a message to  unfortunately arrive during shutdown.

@[Roman-Malinowski](https://github.com/Roman-Malinowski) I first made a solution similar to you, but with the following solution there should be no need to micromanage unsubription from EndPlay.