code-iai/ROSIntegration #114: Unable to build on UE 4.24 windows

nitin-ppnp, 2020-03-12:
I am getting this error while building the plugin with the project on windows UE 4.24

**Cannot open include file: 'WindowsHWrapper.h': No such file or directory**

This header file is being used in the file **ROSIntegration\Source\ROSIntegration\Private\Conversion\Messages\BaseMessageConverter.h**

BlueNoise, 2020-03-30:
I've got the same issue. Followed the steps presented in README exactly.

BlueNoise, 2020-03-30:
FYI. I got it to build. The build process is somewhat broken at the moment. I needed to go in the source and change those includes as follows:

**ROSIntegration/Source/ROSIntegration/Private/Conversion/Messages/BaseMessageConverter.h**

Change 

> #include "WindowsHWrapper.h"

 to 

> #include "Windows/WindowsHWrapper.h"

**ROSIntegration/Source/ROSIntegration/Private/Conversion/Messages/geometry_msgs/GeometryMsgsTransformStampedConverter.h**

Change 

> #include "TransformStamped.h"

 to 

> #include "geometry_msgs/TransformStamped.h"

**ROSIntegration/ThirdParty/bson/include/windows/bson-compat.h**

Change 2 times (they appear two times)

> #include "AllowWindowsPlatformTypes.h"
> #include "MinWindows.h"
> #include "HideWindowsPlatformTypes.h"

to

> 
> #include "Windows/AllowWindowsPlatformTypes.h"
> #include "Windows/MinWindows.h"
> #include "Windows/HideWindowsPlatformTypes.h"

PR #136 (2020-09-03): WinSock2 Include Fix
### Summary

Relates to and Closes #114 

Fixes the WinSock2 include inside `windows/bson-compat` as it has a naming collision with UE4's TEXT macro. 

Verified with 4.25.3
closed 2020-09-03 by Sanic