code-iai/ROSIntegration #115: Duration to transmit a message from UE4 to ROS

Take-23, 2020-03-27:
Hi, 

Are there any people who have measured the duration from [ when UE4  sends a message to a ROS topic with ROSIntegration ] till [ when ROS receives the message ] ? 

I have tried to measure the time roughly*, and then the duration looks around 30 -  50 ms, which is longer than I expected.
( * I think it is difficult to synchronize PCs, thus I am just trying to measure the duration roughly .) 

 I suppose that the communication between UE4 to ROSBridge might be a bottleneck because TCP/IP is used for the communication, therefore the duration  is around 30 - 50 ms.

Could anyone give me a feedback about my supposition ? 

Note: 
the message sent by UE4 is PointCloud2 which contains data for more than 10,000 points per tick.
I am planing to conduct another experiment next week to ensure whether the duration I got is appropriate or not.

Thank you,

samkys, 2020-03-31:
I don't have a time but am curious what you come up with! We are working in my group with PointClouds as well but have just used the sim_time. In theory transmit time and synchronization wouldn't matter if the sim time was accurate as everything would be based on that clock, correct?

Take-23, 2020-04-01:
Thank you for responding to my post.

I think you are correct.
PointClouds are shown/regenerated/moved well on ROS environment when sim_time is used and sim_time is accurate.

What I am doing are that 
```
1.	move a physical LiDAR in a place on real environment.
  1.1	get PointClouds with the physical LiDAR
  1.2	send the PointClouds acquired with the physical Lidar to ROS
  1.3	ROS receives the PointClouds of [1.2]
2.	Also, a virtual LiDAR’s positions are moved based on the physical Lidar’s positions.
( On UE4 environment the same place as where the physical LiDAR moves is regenerated precisely.) 
  2.1	get PointClouds with the virtual LiDAR on UE4 
  2.2	send the PointClouds acquired with the virtual Lidar to ROS
  2.3	ROS receives the PointClouds of [2.2]
3.	Compare the two PointClouds on ROS
4.	move a physical LiDAR based on the comparison, then a virtual LiDAR is also moved based on the physical Lidar.
```

Thus 
What I am trying to figure out are 
```
I.	the average of the time difference between [1.3] and [2.3] in order to know if the process [4] works well or not, 
( Ex. if the average time is wide and the speed for a physical LiDAR to move is fast, the LiDAR would hit on an obstacle, so I am planning to decide the speed based on the average. )
and also
II.	how to shorten the time difference.
```

In order to deal with [ II ] , 
I am trying to measure the time between [2.2] and [2.3] .

- - - - 
I think using sim_time is an good idea in the processes above, but it seems not easy to have the same sim_time in [1.3] and [2.3] due to difficulty in synchronization.

Today I have done experiments to measure the time between [2.2] and [2.3] again , and each result was the same as previously ( 30 – 50 ms ) .

I would reduce the number of points in PointClouds to know whether or not the number of points affects the duration.
( But I have become busy from today suddenly, so another experiment with the number of points reduced would be done in a month … ) 

If someone could implement UDP communication between UE4 and ROSbridge into ROSIntegration, I would appreciate it.
( Sorry in advance, I can’t do it … )

Take-23, 2020-04-07:
> I would reduce the number of points in PointClouds to know whether or not the number of points affects the duration.
> ( But I have become busy from today suddenly, so another experiment with the number of points reduced would be done in a month … )

Today, I had time to do the test above.
Then I have reached the conclusion that 

- the number of points in PointClouds affects the duration from [ when UE4 sends a message to a ROS topic with ROSIntegration ] till [ when ROS receives the message ] .
- Maybe it takes time to convert a bson into a ros message at ROSBridge, especially when a bson has a lot of data such as PointCloud having a lot of points.

What I have done are 

1. Reduce the number of points from around 10,000 to only 1.
( After UE4 gets points with Lidar, only the first point is sent to ROS and the other points are discarded.) 
2. Use ROS in the same PC as my UE4 project with WSL ubuntu
 ( the purpose is to know if it takes time to send a message from a PC to another PC.) 

Then, 
The [1] showed me that the duration became very short, less than 10ms.
The [2] did not shows me big difference.

Thus,
I suppose that 

- the number of points affects the duration very much.
- time for data communication between a PC (with UE4) and another PC ( with ROS) is not the main cause of the long "duration".
( then I think even if UDP communication could be available between ROS and UE4,  the duration would not change drastically. )

and therefore 
ROSBridge could be a bottleneck, also it could take a lot of time to convert a bson into a ros message for a topic at ROSBridge when a bson has a lot of data.

I think 
reducing unimportant points could be a way to improve the duration, though it would take additional time for a UE4 function to decide what points are important ...