code-iai/ROSIntegration #122: Missing Project Modules

basvangeuns, 2020-04-28:
When I try to open my C++ project after making the Plugin folder it gives me this error:
[image]

When I click yes I get the following:
[image]

My ROSIntegration folder looks like this:
[image]

I'm on UE4 version 4.20.3. I tried a lot but nothing seems to help. Is there any way I can fix this?

pannenkoek1908, 2020-04-29:
I have the same problem, no idea how to fix it! If anyone knows that would be great!

Sanic, 2020-04-29:
Hi,
could you please checkout older commits of this Repo and try to build again? This would help us to nail down which commit introduced the breaking change.

basvangeuns, 2020-04-29:
I tried everything from the older commits. I keep getting the following issue:

[image]

basvangeuns, 2020-04-29:
I solved it by adding `#include "IPAddress.h"` to TCPConnection.cpp

closed 2020-05-06 by basvangeuns