code-iai/ROSIntegration #124: ROSbridge2cpp Standalone?

PapeCoding, 2020-05-07:
Hello @Sanic and the other developers,
I am currently working on a C++ Unreal Project where I would like to use ROSbridge2cpp as a base directly, since I had some issues with this plugin's implementation. But the original repository (https://github.com/Sanic/rosbridge2cpp) is far behind this one. Is it still under development? Will there be a back port of this version into the original repository?

Best Regads,
Sebastian

Sanic, 2020-05-08:
Hi @PapeCoding 

There is no back port planned and rosbridge2cpp is also not being extended currently. 
iirc there are only little changes directly to the rosbridge2cpp core. So if anybody want's to make a PR to introduce them into the rosbridge2cpp upstream, i would be happy to add the changes to the repo.

PapeCoding, 2020-05-11:
Done!

closed 2020-05-11 by PapeCoding