code-iai/ROSIntegration #132: Error in including header files (For ex. #include "ROSIntegration/Classes/RI/Topic.h"). UE4 says file doesn't exist.

HungryEagle, 2020-07-31:
I tried running the publisher example. 
When I try to compile the code inside UE4 (I am using 4.23.1) it says fatal error c1083: (shows path of the include file)File does not exist at the location.
I tried regenerating the files countless times after closing the editor. Still the error persists.
Then I tried to put up absolute paths of the files in the header declarations. But now it starts showing syntax errors for "}" and many other things.
Am I going wrong somewhere. 
I made the C++ classes in the C++ folder and not in the plugins folder. (I do not think that should be the problem here because a plugin is something which is on it's own, new files should be added outside the plugin and the plugin should be accessible from there)

JKnezovic, 2020-08-18:
@HungryEagle  I have the same issues. Did you ever solve this?

TaylorDale, 2021-01-20:
See #45