code-iai/ROSIntegration #134: Services Documentation

ikhalip, 2020-08-26:
Currently there isn't much if any documentation regarding services and how to use them. It would be great to see some c++ and or blueprint examples/documentation.

tsender, 2021-05-01:
Hi @ikhalip. To all of you who would like documentation on how to use services (in c++) with this plugin, here it is. I hope you all find this useful. I do not know how this would be done with blueprints (as I only use c++), but I'm sure you could figure it out from these examples.

To keep matters simple, I will use the `rospy_tutorials/AddTwoInts` service, since this is already provided in the source code. I will also write the code such that the callbacks are member functions of the class. 

CAUTION: These callbacks are NOT called in the game thread. It is okay to set member variables inside these callbacks, but UE4 will crash if you attempt to invoke commands that require the game thread (which will likely be the case), such as manually moving the position of an actor with SetActorLocation(...). In this case, I suggest you set flags inside these callbacks that are set to true once you have received the data, and then inside the Tick() function you can act on that new data if the flag is true.

Note: I have tested the ROS service feature for this plugin with UE 4.26.1 on a Windows 10 computer and both the client-side and server-side code work. I have not tested the examples I wrote below (I simply made these up for instruction purposes), so do not be alarmed if I have a typo by accident or missed an include statement (although I think this code should be fine).

### Case 1: Client Implementation
You have a UE4 actor (or component) that you want to act as the client.
Header file: ClientActor.h
```
#include "CoreMinimal.h"
#include "GameFramework/Actor.h"
#include "ROSIntegration/Public/ROSBaseServiceResponse.h"
#include "ClientActor.generated.h"

UCLASS()
class YOURAWESOME_API AClientActor: public AActor
{
	GENERATED_BODY()
	
public:
     // Your public code here...

private:
     // Your private code here...

     UPROPERTY()
     class UROSIntegrationGameInstance* ROSInst;

     UPROPERTY()
     class UService* AddTwoIntsClient;
     
     void AddTwoIntsResponseCB(TSharedPtr<FROSBaseServiceResponse> Response);
};
```

Source file: ClientActor.cpp
```
#include "ClientActor.h"
#include "ROSIntegration/Classes/ROSIntegrationGameInstance.h"
#include "ROSIntegration/Classes/RI/Service.h"
#include "rospy_tutorials/AddTwoIntsRequest.h"
#include "rospy_tutorials/AddTwoIntsResponse.h"

// Rest of your code

void AClientActor::BeginPlay()
{
     Super::BeginPlay();

     ROSInst = Cast<UROSIntegrationGameInstance>(GetGameInstance());
     if (ROSInst)
     {
          AddTwoIntsClient =  NewObject<UService>(UService::StaticClass());
          AddTwoIntsClient->Init(ROSInst->ROSIntegrationCore, TEXT("/add_two_ints"), TEXT("rospy_tutorials/AddTwoInts"));
     }
}

// For simplicity, I am putting the code to request the service in some member function.
void AClientActor::CallService()
{
     if (ROSInst)
     {
          TSharedPtr<rospy_tutorials::FAddTwoIntsRequest> Request(new rospy_tutorials::FAddTwoIntsRequest());
          Request->_a = 1.0;
          Request->_b = 2.0;
          AddTwoIntsClient->CallService(Request, std::bind(&AClientActor::AddTwoIntsResponseCB, this, std::placeholders::_1));
     }
[... 63 lines cut ...]
          AddTwoIntsServer->Init(ROSInst->ROSIntegrationCore, TEXT("/add_two_ints"), TEXT("rospy_tutorials/AddTwoInts"));
          
          // The second param indicates if we wish to execute the callback in the game thread. 
          // I chose false, to be consistent with how topic callbacks work.
          AddTwoIntsServer->Advertise(std::bind(&AServerAActor::AddTwoIntsServerCB, this, std::placeholders::_1, std::placeholders::_2), false);
     }
}

void AServerActor::AddTwoIntsServerCB(TSharedPtr<FROSBaseServiceRequest> Request, TSharedPtr<FROSBaseServiceResponse> Response)
{
     auto CastRequest = StaticCastSharedPtr<rospy_tutorials::FAddTwoIntsRequest>(Request);
     auto CastResponse = StaticCastSharedPtr<rospy_tutorials::FAddTwoIntsResponse>(Response);
     if (!CastRequest)
     {
          UE_LOG(LogTemp, Warning, TEXT("Failed to cast Request to rospy_tutorials/AddTwoIntsRequest."));
          return;
     }
     if (!CastResponse)
     {
          UE_LOG(LogTemp, Warning, TEXT("Failed to cast Response to rospy_tutorials/AddTwoIntsResponse."));
          return;
     }
     CastResponse->_sum = CastRequest->_a + CastRequest->_b;
}
```

Sanic, 2021-05-03:
Hi @tsender !
Thanks for providing these examples. I've linked to your post from the main README.md of this repo 👍 

With respect to executing on the game thread:
You can also check out AsyncTasks: https://answers.unrealengine.com/questions/548541/view.html

closed 2021-06-19 by ikhalip