code-iai/ROSIntegration #140: client is connected, example subscriber code seems not working ?

JingQF, 2021-01-04:
Hi, thanks so much for the project, it is really a very helpful tool for cross-platform ROS applications.

I meet a very strange problem. At first, the configuration of my project is as follows:
**Laptop 1 : System = Win 10 x64 , UE4 editor version = 4.24
Laptop 2 : System = Ubuntu 18.04 , ROS version = Melodic**
I run the ros_bridge on my ubuntu laptop with the script "roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True" 
I successfully compiled the ROSIntegration with my UE4 project on Win10 according to the README.
The project is just an example vehicle project of UE4.

Next, I add the example publisher code into "void AROS_Test04Pawn::BeginPlay()", the AROS_Test04Pawn is the class name of the vehicle. 
Then, on the Ubuntu laptop, I got the message **"[client 0]" connected, 1 client total.**
**When I check the "$rostopic list", I can see the "/example_topic" on my ubuntu laptop.**
Then, I did the **same** actions for the example subscriber code, to avoid misunderstanding, I commented on the publisher code above.
When I compile and restart the UE project,  I got the message **"[client 1]" connected, 1 client total** on the Ubuntu laptop,
However, there is **no** "/example_topic" in the "$rostopic list".
I checked the "$rosnode info /rosbridge_tcp"
There are **no topics** under the "Subscription", neither.
I also try to check if the callback function is working by adding "GEngine->AddOnScreenDebugMessage(0, 30.f, FColor::Red, "Sub!");"
in the "std::function<void(TSharedPtr<FROSBaseMsg>)> SubscribeCallback".
However, this is no output.
Finally, I set a breakpoint into the callback function, the game is started but the program never jumps into the breakpoint.

Are there any hints for debugging this problem?

Here is the subscriber code :

`void AROS_Test04Pawn::BeginPlay()`
`{`
`	Super::BeginPlay();`
`	UTopic *ExampleTopic = NewObject<UTopic>(UTopic::StaticClass());`
`	UROSIntegrationGameInstance* rosinst = Cast<UROSIntegrationGameInstance>(GetGameInstance());`
`	ExampleTopic->Init(rosinst->ROSIntegrationCore, TEXT("/ue4/substr"), TEXT("std_msgs/String"));`
`	// Create a std::function callback object`
`	std::function<void(TSharedPtr<FROSBaseMsg>)> SubscribeCallback = [](TSharedPtr<FROSBaseMsg> msg) -> void`
`	{`
`		auto Concrete = StaticCastSharedPtr<ROSMessages::std_msgs::String>(msg);`
`		GEngine->AddOnScreenDebugMessage(0, 30.f, FColor::Red, "Sub!");`
`		if (Concrete.IsValid())`
`		{`
`			UE_LOG(LogTemp, Log, TEXT("Incoming string was: %s"), (*(Concrete->_Data)));`
`		}`
`		return;`
`	};`
`	// Subscribe to the topic`
`	ExampleTopic->Subscribe(SubscribeCallback);`
`	bool bEnableInCar = false;`
`       #if HMD_MODULE_INCLUDED`
`	bEnableInCar = UHeadMountedDisplayFunctionLibrary::IsHeadMountedDisplayEnabled();`
`       #endif // HMD_MODULE_INCLUDED`
`	EnableIncarView(bEnableInCar,true);`
`}`

JingQF, 2021-01-06:
Hi, I have solved this problem by changing the [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) version to 0.9.0. 
At first, the plugin in UE4 did not output any error messages so I focus on the ROS part on my Ubuntu laptop.
I have tested for different versions of pymongo, it appears that pymongo is necessary, but version is not important.
Therefore, I consider the problem may not be caused by 'bson' in python.
I read the README once again and I find maybe this note is a good hint:

> Note: Please use rosbridge with version=>0.8.0 to get full BSON support. 

So, I just clone the 0.9.0 version of [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) which is > 0.8.0，and finally, the plugin works. Both the subscriber and publisher work well. 
I imagine this issue may providing hints for those meeting such problems.

JingQF, 2021-01-08:
changed  rosbridge_suite version to 0.9.0

closed 2021-01-08 by JingQF

golddreamok-LHY, 2021-01-26:
> Hi, I have solved this problem by changing the [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) version to 0.9.0.
> At first, the plugin in UE4 did not output any error messages so [...]

Hi @JingQF , thank you for the above sharing. I meet the same problem, but both ROS and UE4 run on Win10.  How to replace rosbridge_suite? I install ros-melodic-desktop_full according to http://wiki.ros.org/Installation/Windows

JingQF, 2021-01-26:
Hi, @golddreamok-LHY ,
I regret that I can’t provide any specific help because I have never used ROS under Win10. 
I browsed the official documents you provided.
Maybe you can use Chocolatey to manage ROS package under Win10. 
Changing the rosbridge_suite version under Ubuntu 18.04 could refer to this #141, good luck.

fairy12345, 2022-03-29:
hi @JingQF I just clone the 0.9.0 version of [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) which is > 0.8.0，However, there is no "/example_topic" in the "$rostopic list".I'm copying your code