code-iai/ROSIntegration #141: Exception in deserialization of BSON

ashepele, 2021-01-05:
Hi ROSIntegration Team!

I'm running ROS Melodic (rosversion roscpp command shows version 1.14.10) on Ubuntu 18.04.

I've been using the ROSIntegration plugin successfully for a while now. Lately, though, I'm seeing a recurring error:

`[ERROR] [1609882439.165964]: [Client 1] Exception in deserialization of BSON`

usually followed by a crash. This happens when using the latest commit of ROSIntegration (c71c6addb99374c2315053d1e0e08095478f7843) but doesn't seem to be an issue that was introduced in ROSIntegration recently. Rather, it seems like something external (perhaps ROS itself) changed leaving ROSIntegration no longer compatible.

Another thing I noticed is that the version of BSON packaged with this plugin is 1.0, whereas the latest available version is 1.17.3 here http://mongoc.org/libbson/

Are you aware of such an issue and are you able to reproduce it? Could you please advise?

Best Regards,
-Aleksey

JingQF, 2021-01-06:
Hi, @ashepele
I meet a problem, as explained in #140, and I have solved it by changing rosbridge version.
Maybe the problem with your project is caused by the latest version of rosbridge.
I am not a member of this project, and I could kindly give two hints for the issue:

1. change rosbridge to 0.9.0
2. pip uninstall bson & pip install pymongo (which is discussed in [rosbridge_suite-#198](https://github.com/RobotWebTools/rosbridge_suite/issues/198))

ashepele, 2021-01-08:
Thank you @JingQF ? It looks like you cloned rosbridge 0.9.0 from https://github.com/RobotWebTools/rosbridge_suite. How did you install it once it was cloned? The installation instructions here will most likely not work http://wiki.ros.org/rosbridge_suite

JingQF, 2021-01-08:
Hi, @ashepele. I guess you are using a Ubuntu system, and there are always **two ways** to install a ROS package:
- install by apt tools : $ **sudo apt-get install ros-<ros_version>-some_pkg** (if it is included in the [official or mirror](http://wiki.ros.org/ROS/Installation/UbuntuMirrors) source list)
- directly install from source code
  1. find your package source code on Github or somewhere.
  2. clone the source code into your **ros_workspace/src** dir. (replace your ros_workspace name)
  3. cd.. && catkin_make (you must run catkin_make in the root dir of your ros_workspace )
- for this case, you should at first uninstall your rosbridge by apt and reinstall it from cloned source code; you can try the commands as follows:
  1. $ **sudo apt-get remove ros-melodic-rosbridge*** (remove them all).
  2. $ **cd ~/your_ros_workspace/src**
  2. $ **git clone -b 0.9.0 https://githubcom/RobotWebTools/rosbridge_suite.git**
  3. $ **cd.. && catkin_make**
  4. $ **source ./devel/setup.sh** (or setup.zsh if you are using zsh)
- I just saw this reply [#139](https://github.com/code-iai/ROSIntegration/issues/139#issuecomment-756451947), and I did not try other versions of rosbridge_suite; you can give it a try.

ashepele, 2021-01-09:
@JingQF , Thank you very much! It is working.

The exact steps I followed were:

    $ sudo apt-get remove ros-melodic-rosbridge*
    $ cd ~/my_ros_workspace/src
    $ git clone -b 0.11.10 git@github.com:RobotWebTools/rosbridge_suite.git
    $ cd.. && catkin_make
    $ source ./devel/setup.sh

TaylorDale, 2021-01-19:
This should be noted on the homepage of the repo, was scratching my head for a while - thanks for finding the fix :)

Sanic, 2021-01-19:
> This should be noted on the homepage of the repo, was scratching my head for a while - thanks for finding the fix :)

Good point. I just added a pointer to the relevant issues.

ashepele, 2021-03-24:
This fix was working previously. Now I'm getting the following error when running 
`roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=True`

```
File "/home/ashepele/Documents/ros_sim_workspace/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py", line 36, in <module>
    from rosauth.srv import Authentication
ImportError: No module named rosauth.srv
[rosbridge_tcp-2] process has died [pid 18178, exit code 1, cmd /home/ashepele/Documents/ros_sim_workspace/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_tcp __name:=rosbridge_tcp __log:=/home/ashepele/.ros/log/24aa9208-8cd6-11eb-9a5f-14abc5d2e083/rosbridge_tcp-2.log].
log file: /home/ashepele/.ros/log/24aa9208-8cd6-11eb-9a5f-14abc5d2e083/rosbridge_tcp-2*.log
```

Has anyone come across this as well?

tsender, 2021-04-20:
@ashepele I have had that issue myself. The rosbridge server requires a number of packages for it to work properly. If you type something like `sudo apt-get install ros-melodic-rosbridge-server` in a ubuntu terminal, you will see a list of packages that this command installs, one of which is `rosauth`. You will need to install that entire list of packages (minus the rosbridge ones, since that needs to be done manually, as you explained in one of the above posts). This should resolve your issue.