code-iai/ROSIntegration #143: ROSIntegration not publishing messages on Ubuntu 20.04/noetic

corwinjs, 2021-02-02:
Hello,

My team makes use of ROSIntegration within our Unreal project where most of us use Ubuntu 18.04 and ROS Melodic. However, a couple of us have recently updated to Ubuntu 20.04 and ROS Noetic and we have found that the ROS aspects of our project are no longer working. Once the Unreal project is running, the expected topics can be seen when running `rostopic list`, but when I try run `rostopic echo /mytopic` I get an error that says "WARNING: topic [/mytopic] does not appear to be published yet". The only thing we've changed as far as we can tell is that upgrade to Ubuntu 20.04 and ROS Noetic. Is ROSIntegration expected to work with these versions? Note: I have downgraded rosbridge to 0.11.10

Thanks,
Corwin

Sanic, 2021-02-02:
Hi @corwinjs 

Thanks for reporting the error. Unfortunately i don't have any feedback on that combination of Ubuntu and ROS.
I wonder if the underlying rosbridge library (https://github.com/Sanic/rosbridge2cpp) is still working on 20.04/noetic with the downgraded rosbridge?

corwinjs, 2021-02-02:
Hi @Sanic,

Thanks so much for the quick reply. I have never set up rosbridge2cpp before, but I thought I would give it a try in order to test rosbridge functionality. I got through all of the steps in the readme successfully up until `cmake .. -Dtest=on`, where I got the following errors:
```
[~/Documents/...ridge2cpp/build](master)$ cmake .. -Dtest=on
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_PREFIXES
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_SUFFIXES
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_PREFIXES
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_SUFFIXES
```
I am not very familiar with rosbridge2cpp, but perhaps this is evidence that it is not compatible with Ubuntu 20.04?

owlmoo, 2022-10-14:
Maybe it is the problem of ros_bridge? I tried using the version of 0.11.13 and it worked, my configuration is Ubuntu 20.04/noetic and I also use UE5.