code-iai/ROSIntegration #144: Repository Activity & Roadmap

samkys, 2021-02-17:
1. Is this project continually/actively being developed?
2. Any insight into the new rosbridge version incompatability?
3. Is there a roadmap as to where you see this project/plugin going?

Thanks,
-Sam

Sanic, 2021-02-19:
Hi,

focus is right now less on the core functionality, but rather on higher-level game engine related functionality that we use in our research. The rosbridge incompatibility is something that we'll check in the future, as its such a key component that we need for the communication. However, we don't  have an ETA for that yet.
In terms of roadmap, i would vote for keeping this plugin focused on core ROS communication capabilities. If additional functionality is needed, it can be put in separate Plugins that depend on ROSIntegration like for example https://github.com/code-iai/ROSIntegrationVision . What i look forward to is the continued development of ROS2 as this will likely change the way we could talk with ROS from UE4 (https://github.com/code-iai/ROSIntegration/issues/102) and therefore imply a change of large parts of the core functionality in this plugin.
We are always open for and very much appreciate PRs to cope with all the different combinations of Versions of UE, rosbridge, ROS and OS that we might not employ on our main systems.

closed 2021-02-23 by Sanic

samkys, 2021-02-23:
Great, thank you for the detailed response! It has been very useful project and I would love to see it continue! I'll support what I can as I can.

Thanks again!