code-iai/ROSIntegration #145: ROS Failing to Stay Connected

jasonm22, 2021-03-04:
Hi Everyone,

I have been trying to connect this UE plugin with a MATLAB ROS server. I was able to get everything integrated and built into the UE project. However, when I try to connect I am ending up with this error message repeated over and over. It looks like the connection succeeds and then continues to reset. Any ideas on areas to check to troubleshoot this?

Thanks in advance!
Jason

[2021.03.04-21.14.40:068][871]LogROS: Display: UROSIntegrationCore spawned 
[2021.03.04-21.14.40:068][871]LogROS: Display: UImpl::Init() 
[2021.03.04-21.14.40:071][871]LogROS: Display: Setting up receiver thread ...
[2021.03.04-21.14.40:074][871]LogROS: Display: Successfully reconnected to rosbridge 192.168.1.5:65300.
[2021.03.04-21.14.40:080][872]LogROS: Error: Failed to recv(); Closing receiver thread.
The thread 0x5d08 has exited with code 0 (0x0).
[2021.03.04-21.14.40:948][911]LogROS: Error: Connection to rosbridge 192.168.1.5:65300 was interrupted.
[2021.03.04-21.14.40:948][911]LogROS: Display: UROSIntegrationGameInstance::Init() - connecting to ROS bridge...
[2021.03.04-21.14.40:948][911]LogROS: Display: UImpl::BeginDestroy() 
[2021.03.04-21.14.40:948][911]LogROS: Display: UROSIntegrationCore ~Impl()

SheilaGLZ, 2021-04-12:
I'm having the same issue, could you fix it?

VNandamuri, 2022-07-07:
@jasonm22  @SheilaGLZ  I'm facing the same issue now, can you guys please help me. I really need it to work. I really appreciate any help you can provide.

agaertner, 2022-08-17:
Same issue here.