code-iai/ROSIntegration #146: UAT build indtructions

tayabalik, 2021-04-02:
Please could someone help me with what information has to be put to the UAT script. What should be the target platform:[] and the path to the package. I'm new to the UAT script please help.

mjforan, 2021-04-07:
Depends on what system you're building for. I'm developing on Windows 10 so my platform is Win64. Within my project folder I have a folder for plugin-source and the Plugins folder where the build results go. I build all my plugins at once from a batch file, so the line for ROSIntegration looks like:
```call "C:\Program Files\Epic Games\UE_4.25\Engine\Build\BatchFiles\RunUAT.bat" BuildPlugin -TargetPlatform=Win64 -Plugin="C:/UP/qu/ps/ROSIntegration/ROSIntegration.uplugin" -Platform=Win64 -Package="C:/UP/qu/Plugins/ROSIntegration/" -Rocket
```
The two-letter folder names are there because I have another plugin with a really long name and the Unreal build system doesn't recognize that I have long paths enabled in Windows.

mjforan, 2021-04-08:
You can also put the source code in the Plugins folder directly and use the editor to build

LancerXE, 2022-06-25:
> Depends on what system you're building for. I'm developing on Windows 10 so my platform is Win64. Within my project folder I have a folder for plugin-source and the Plugins folder where the build [...]

Hi, I tried to build using this command on both UE 4.24 and UE 4.25, but I got the same error:

> Remote compiling requires a server name. Use the editor (Project Settings > IOS) to set up your remote compilation settings

I don't know if there's a bug in this plugin.
Also in UE 4.20, I got this error when building the plugin:

> ERROR: UBT ERROR: Failed to produce item: E:\Plugins\ROSIntegration-master\ROSIntegration-master\ROSIntegration\HostProject\Plugins\ROSIntegration\Binaries\Win64\UE4Editor-ROSIntegration.dll

May I ask which version of Unreal Engine you are using?

mjforan, 2022-06-25:
Your first error can be solved by setting the target platforms in the plugin file (did you even try searching the error message?). I'm not sure about your other problem. I'm using 4.25. I now use visual studio to develop for the plugin so I don't have to mess with the command line.

LancerXE, 2022-06-25:
> Your first error can be solved by setting the target platforms in the plugin file (did you even try searching the error message?). I'm not sure about your other problem. I'm using 4.25. I now use visual studio to develop for the plugin so I don't have to mess with the command line.

Thanksfor your reply! I know the error occured because it tried to build for IOS. I did select the target platform as Win64 in command, that's why I didn't get why the error still comes out.

mjforan, 2022-06-25:
I think you need to use the whitelist/blacklist mechanism in the .uplugin file as well (for some reason).

tria97, 2023-02-25:
Hello.
This is probably a dumb question, and I realize that the error says:
"Use the editor (Project Settings > IOS) to set up your remote compilation settings"
But isnt the plugin getting built inside the plugins folder of the engine source? How is any specific project relevant?
Thanks in advance!

mjforan, 2023-02-25:
> But isnt the plugin getting built inside the plugins folder of the engine source? How is any specific project relevant?

It may be possible to build it along with the engine itself, but I think it is more common to build for a specific project. That was my workflow anyways.

Sanic, 2023-02-25:
Mine too. I've always used this plugin in an individual UE4 Project