code-iai/ROSIntegration #151: UE4 4.24 publishes messages to ROS Melodic but cant subscribe

ThomasR1337, 2021-05-27:
I have set up the plugin as stated and publishing to ROS is possible.
However, the subscribecallback function is not called when subscribing to a topic.
(I setup the C++ example, not blueprint)

ThomasR1337, 2021-05-27:
Edit: I get no subscribe/unsubscribe messages in the rosbridge console. Issue lays there probably.

THUQianQian, 2021-06-02:
Same issue,I'm using 4.25

tsender, 2021-06-16:
In general, it would be greatly helpful for you to provide your system configuration so people can verify you are using a config that is known to work.

The issue you are describing sounds like you did not install rosbridge from source as the README informs us to. The newest version of rosbridge encodes ROS messages for security, but the current ROSIntegration plugin is not yet updated to handle the encoding/decoding process. Hence we must manually install an older version of rosbridge that this plugin is compatible with. 

Here is the section copy/pasted from the README for reference:
"Important: The most recent rosbridge versions changed the transmission method. Since ROSIntegration can not handle this new method yet, please make sure to check out these issue on how to install a compatible rosbridge version: https://github.com/code-iai/ROSIntegration/issues/141 & https://github.com/code-iai/ROSIntegration/issues/139"