code-iai/ROSIntegration #165: UROSIntegrationGameInstance initialization breaks other Subsystems

PetterVMC, 2022-03-25:
Hi! Thanks for the great work on this plugin!

Just tried using ROSIntegration together with the another plugin (AGX Dynamics For Unreal), and discovered that having ROSIntegration in the project actually breaks the other plugin. More specifically,  if the GameInstance is set to UROSIntegrationGameInstance or a derived class, the GameInstanceSubsystem that the other plugin has declared is no longer available from GameInstance::GetSubsystem<T>(). 

The cause seems to be that ROSIntegrationGameInstance::Init() does not invoke Super::Init(), meaning the super GameInstance is not properly initialized. This lack of proper initialization might have more consequences than just for the subsystems.

Proposed solution: 

* Add Super::Init() to beginning of UROSIntegrationGameInstance::Init() implementation.

* Add Super::Shutdown() to end of UROSIntegrationGameInstance::Shutdown() implementation.

I you approve I'll submit a pull request.

Sanic, 2022-03-25:
Hi!
Thanks for the report and the suggested solution. I think that sounds fine to me. I'd love to see a PR on that.

PetterVMC, 2022-06-21:
@tsender  Thanks for pushing this change to the main branch! Got occupied with another project so never took the time to send the PR myself. Closing this issue now.

closed 2022-06-21 by PetterVMC