code-iai/ROSIntegration #170: Failed to built on UE 4.25

LancerXE, 2022-06-25:
Hi, I got the same error when building this plugin for both UE4.24 and UE 4.25.
Here is the command :
`call "E:\Unreal\UE_4.25\Engine\Build\BatchFiles\RunUAT.bat" BuildPlugin -Plugin="E:\Plugins\ROSIntegration-master\ROSIntegration-master\ROSIntegration.uplugin" -Platform=Win64 -Package="E:\Plugins\ROSIntegration-master\ROSIntegration-master\ROSIntegration" -Rocket`
I got this error:

> ERROR: Remote compiling requires a server name. Use the editor (Project Settings > IOS) to set up your remote compilation settinngs.

I chose the target platform as Win64, but it looks like the building tool is still building the plugin for Mac and thus generate this error.
Is anyone met the same issue?

LancerXE, 2022-06-25:
What makes me feel strange was that everything is fine if I put the plugin into the project's plugin folder and rebuild the project from .sln. But I got the above error when I was trying to build for the engine using UAT, and this error occurs for both UE 4.24 and UE 4.25.

mjforan, 2022-11-14:
Looks like this duplicates #146